autorope/donkeycar #370: Where is manage.py in this repository?

RossMelbourne, 2019-01-29:
I am new to Donkey Car and still learning GitHub.  I am not able to find manage.py even though I can find it installed in the mycar folder on the Pi. I was able to copy it from the Pi to my Mac using a scp command, just to learn how it works.  I was expecting to be able to find the file when I did a search of this repository.

scitronboy, 2019-01-30:
https://github.com/autorope/donkeycar/blob/dev/donkeycar/templates/donkey2.py is the manage.py code.

The script that generates `manage.py` is in the base script, in `createcar`:

https://github.com/autorope/donkeycar/blob/dev/donkeycar/management/base.py

RossMelbourne, 2019-01-31:
Thank you for pointing that out.  I would suggest a comment in the code for people searching for manage.py so they find the comment in the code as well as this thread.  Now I am curious about the advantages of copying and renaming code files.  I find it amazing how flexible python is and what people do with it.

tawnkramer, 2019-06-11:
I will make a note too add more documentation about how templates are used and become the manage.py when copied to the user folder.

closed 2019-06-11 by tawnkramer