autorope/donkeycar #378: [Errno 121] Remote I/O error

rongfeng-china, 2019-03-21:
Hello, I am trying to execute the donkey calibrate --channel 0 command but am getting the error: [Errno 121] Remote I/O error...

My car used to work, but this time it stop working. The I2C is enabled, and the wires seem to connect in the right way. Any idea? How to fix it? Thanks!

879099766, 2019-03-29:
According to doug on Slack, "make sure you have the I2C wires from the rPi to the PCA9685 correctly installed. The docs show pictures and not pin labels so you have to count the pins correctly and get the correct side of the 40 pin connector.  The I2C interface is 3.3V so please do not connect the 5V pins."

Link: https://donkeycar.slack.com/archives/C4HR56WN6/p1537679984000100?thread_ts=1537679739.000100&cid=C4HR56WN6

So, please search the issue on Slack before asking because it will save you lots of time. :)

Good luck with the troubleshooting.

tawnkramer, 2019-06-04:
we could use a FAQ with these common errors. Please let us know if you still need help. closing for now.

closed 2019-06-04 by tawnkramer