autorope/donkeycar #400: makemovie with salient becomes very slow

Schmill, 2019-06-24:
If you use the makemovie command with the --salient argument it starts at a reasonable speed, but just 100 frames in it starts to significantly slow down.
It doesn't take long before it is so slow as to be unusable.
As an example my PC produces the first few frames, performing the "if self.convolution_part:" part of make_frame at 0.6 seconds per frame (estimating that it will complete 200 frames in 1:48 ), and is doing 1.67it/s.
By the time it performs the 200th frame it is already taking 1.14 seconds per frame, and has dropped to 1.15s/it.

I'm not sure _why_ it would be slowing down like this?

dlarue, 2019-07-16:
What operating system are you running on?  Have you looked at your system wide process usage to see if anything else was going on when you noticed longer processing time for salient object visualization generation?

Schmill, 2019-07-16:
This was running on Windows, nothing else was happening, the overall process just slows down.
Similar report here on the slack channel:
https://donkeycar.slack.com/archives/C4B69691N/p1563298532105700

DocGarbanzo, 2019-08-24:
Have a look at PR #450, I think I fixed this.

tawnkramer, 2019-08-25:
Awesome! How cool is that?

Schmill, 2019-08-25:
Excellent!
Not tried it yet, but this is great news!
Thanks DG!

Schmill, 2019-09-04:
All working quickly now - I still get really markup happening, but at least its faster!
Cheers DG!

closed 2019-09-04 by Schmill