autorope/donkeycar #404: Some utilities not respecting default model type from config file

Schmill, 2019-06-25:
I'd been having issues with the new code base until I realised this. _Some_ but not _all_ of the commands / utilities use the default model type that is given in the config file, but some are using their _own_ default. 
As an example, "tubplot" now seems to default to categorical if not specified on the command line, irrespective of what is specified in the config file.
I propose to make a change to use the type in the following order of precedence:
* Command line specified.
* Config file specified

If neither are given then error.

Schmill, 2019-06-25:
Turns out it is probably _only_ the tubplot that has this problem, will do a simple PR later.

tawnkramer, 2019-06-25:
Thanks for fixing. Categorical used to be the default. Now it uses the config setting. Hasn’t been universally applied.

closed 2019-06-25 by tawnkramer

Schmill, 2019-06-25:
No problem. It wasn't so much that it used a default, but that it did it silently and didn't choose the one that was expected :)