autorope/donkeycar #417: Compile Errors in Nano Instructions -

dmccreary, 2019-07-05:
I tried to follow the instructions to compile opencv but got errors here:

http://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/#step-4-install-opencv

I fixed some errors in the instructions but I am still getting errors when I run cmake

Here is the command:

cmake -D CMAKE_BUILD_TYPE=RELEASE \
    -D CMAKE_INSTALL_PREFIX=/usr/local \
    -D INSTALL_PYTHON_EXAMPLES=ON \
    -D INSTALL_C_EXAMPLES=OFF \
    -D OPENCV_ENABLE_NONFREE=ON \
    -D OPENCV_EXTRA_MODULES_PATH=~/projects/cv2/opencv/modules \
    -D PYTHON_EXECUTABLE=~/env/bin/python \
    -D BUILD_EXAMPLES=ON ../opencv

Here are the errors:
$ ./c-make.sh
-- Looking for ccache - not found
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found suitable version "1.2.11", minimum required is "1.2.3") 
-- Could NOT find Jasper (missing: JASPER_LIBRARIES JASPER_INCLUDE_DIR) 
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.11") 
-- Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
-- Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
-- Could NOT find Atlas (missing: Atlas_CLAPACK_INCLUDE_DIR) 
-- A library with BLAS API found.
-- A library with LAPACK API found.
-- Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH) 
-- Could NOT find Pylint (missing: PYLINT_EXECUTABLE) 
-- Could NOT find Flake8 (missing: FLAKE8_EXECUTABLE) 
-- VTK is not found. Please set -DVTK_DIR in CMake to VTK build directory, or to VTK install subdirectory with VTKConfig.cmake file
 > /home/dan/projects/cv2/opencv/modules/calib3d
 > /home/dan/projects/cv2/opencv/modules/core
 > /home/dan/projects/cv2/opencv/modules/dnn
 > /home/dan/projects/cv2/opencv/modules/features2d
 > /home/dan/projects/cv2/opencv/modules/flann
 > /home/dan/projects/cv2/opencv/modules/gapi
 > /home/dan/projects/cv2/opencv/modules/highgui
 > /home/dan/projects/cv2/opencv/modules/imgcodecs
 > /home/dan/projects/cv2/opencv/modules/imgproc
 > /home/dan/projects/cv2/opencv/modules/java
 > /home/dan/projects/cv2/opencv/modules/js
 > /home/dan/projects/cv2/opencv/modules/ml
 > /home/dan/projects/cv2/opencv/modules/objdetect
 > /home/dan/projects/cv2/opencv/modules/photo
 > /home/dan/projects/cv2/opencv/modules/python
 > /home/dan/projects/cv2/opencv/modules/stitching
 > /home/dan/projects/cv2/opencv/modules/ts
 > /home/dan/projects/cv2/opencv/modules/video
 > /home/dan/projects/cv2/opencv/modules/videoio
 > /home/dan/projects/cv2/opencv/modules/world
 > /home/dan/projects/cv2/opencv/modules/calib3d
 > /home/dan/projects/cv2/opencv/modules/core
 > /home/dan/projects/cv2/opencv/modules/dnn
 > /home/dan/projects/cv2/opencv/modules/features2d
 > /home/dan/projects/cv2/opencv/modules/flann
 > /home/dan/projects/cv2/opencv/modules/gapi
 > /home/dan/projects/cv2/opencv/modules/highgui
 > /home/dan/projects/cv2/opencv/modules/imgcodecs
 > /home/dan/projects/cv2/opencv/modules/imgproc
 > /home/dan/projects/cv2/opencv/modules/java
 > /home/dan/projects/cv2/opencv/modules/js
 > /home/dan/projects/cv2/opencv/modules/ml
 > /home/dan/projects/cv2/opencv/modules/objdetect
 > /home/dan/projects/cv2/opencv/modules/photo
 > /home/dan/projects/cv2/opencv/modules/python
 > /home/dan/projects/cv2/opencv/modules/stitching
 > /home/dan/projects/cv2/opencv/modules/ts
 > /home/dan/projects/cv2/opencv/modules/video
 > /home/dan/projects/cv2/opencv/modules/videoio
 > /home/dan/projects/cv2/opencv/modules/world
CMake Error at cmake/OpenCVModule.cmake:354 (message):
  Duplicated modules LOCATIONS has been found
Call Stack (most recent call first):
  cmake/OpenCVModule.cmake:371 (_assert_uniqueness)
  modules/CMakeLists.txt:7 (ocv_glob_modules)

-- Configuring incomplete, errors occurred!
See also "/home/dan/projects/cv2/build/CMakeFiles/CMakeOutput.log".
See also "/home/dan/projects/cv2/build/CMakeFiles/CMakeError.log".

I wonder if this line is relevent:

 Could NOT find Jasper (missing: JASPER_LIBRARIES JASPER_INCLUDE_DIR) 
Any suggestions?

Thanks!

tikurahul, 2019-07-05:
I think you are missing `jasper`. Try using the suggestion here:

https://github.com/opencv/opencv/issues/8622#issuecomment-353761271

```
sudo add-apt-repository "deb http://security.ubuntu.com/ubuntu xenial-security main"
sudo apt update
sudo apt install libjasper1 libjasper-dev
```

tawnkramer, 2019-07-14:
Did this solve the issue? Should we add this to docs?

dmccreary, 2019-07-19:
> Did this solve the issue? Should we add this to docs?

No, I am not able to install jasper.

$ sudo add-apt-repository "deb http://security.ubuntu.com/ubuntu xenial-security main"
Get:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]                                                                               
Get:6 file:/var/visionworks-repo  Release [1,999 B]                                                                                      
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                                                                                  
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                                                                                  
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                                                                             
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                                                                             
Hit:9 http://ppa.launchpad.net/jonathonf/python-3.6/ubuntu bionic InRelease                                                              
Hit:10 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                                                                             
Hit:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease   
Get:11 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]
Hit:13 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease        
Hit:15 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease                
Ign:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages         
Get:20 http://security.ubuntu.com/ubuntu xenial-security/main Translation-en [279 kB]
Get:21 http://security.ubuntu.com/ubuntu xenial-security/main arm64 DEP-11 Metadata [67.9 kB]
Get:22 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 48x48 Icons [14.6 kB]
Get:23 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 64x64 Icons [73.2 kB]
Ign:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages
Ign:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages
Err:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages
  404  Not Found [IP: 2001:67c:1360:8001::17 80]
Fetched 109 kB in 4s (29.5 kB/s)                         
Reading package lists... Done
E: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/main/binary-arm64/Packages  404  Not Found [IP: 2001:67c:1360:8001::17 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
(env) dan@nvidia-nano:~/projects/cv2/build$ sudo apt update
Get:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]                                                      
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                                                                        
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                                                                                  
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                                                                             
Hit:9 http://ppa.launchpad.net/jonathonf/python-3.6/ubuntu bionic InRelease                                                              
Hit:10 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                                                                             
Hit:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease   
Get:11 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]
Hit:13 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease            
Hit:14 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease             
Ign:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages
Get:20 http://security.ubuntu.com/ubuntu xenial-security/main Translation-en [279 kB]
Get:21 http://security.ubuntu.com/ubuntu xenial-security/main arm64 DEP-11 Metadata [67.9 kB]
Get:22 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 48x48 Icons [14.6 kB]
Get:23 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 64x64 Icons [73.2 kB]
Ign:19 http://security.ubuntu.com/ubuntu xenial-security/main arm64 Packages
[... cut ...]
E: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/main/binary-arm64/Packages  404  Not Found [IP: 2001:67c:1360:8001::17 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
(env) dan@nvidia-nano:~/projects/cv2/build$ sudo apt install libjasper1 libjasper-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libjasper1
E: Unable to locate package libjasper-dev

dmccreary, 2019-07-19:
I am looking into this fix: https://askubuntu.com/questions/996103/16-04-update-packages-not-found

tikurahul, 2019-07-19:
Are you using the latest SD image from Nvidia ? They made a bunch of updates in May.

dmccreary, 2019-07-21:
No, my image is older and I have been having problems doing an "update" - I will take your advice and download a new image...Thank you Rahul!

dmccreary, 2019-07-22:
I did upgrade to the latest Nvidia disk image and now my libraries do compile OK.  My problem is that I had an old image I downloaded in March and I had only a single 128GB microSD card lots of demos working on it, so I didn't want to wipe out the entire disk image.   Too bad there was no upgrade scirpt!

https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit

dmccreary, 2019-07-22:
We can mark this as closed.  It would be nice to document that you have to use the right image in the instructions.

dmccreary, 2019-07-22:
The instructions assumed you have the right version of the Nivida Nano image.

closed 2019-07-22 by dmccreary