autorope/donkeycar #440: please checks your tub path or provide meta data

python1995, 2019-07-25:
I was trying to train donkeycar

and I get this

(env) pi@raspberrypi:~/donkey2 $ python train.py C:\Users\erdoe.LAPTOP-E8A1PPIN\Desktop\tub
/home/pi/.local/lib/python3.5/site-packages/env/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: compiletime version 3.4 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.5
  return f(*args, **kwds)
/home/pi/.local/lib/python3.5/site-packages/env/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: builtins.type size changed, may indicate binary incompatibility. Expected 432, got 412
  return f(*args, **kwds)
Namespace(base_model=None, batch_size=64, model='model_2019-07-25__15-37.h5', train_split=0.8, tubs=['C:Userserdoe.LAPTOP-E8A1PPINDesktoptub'])
tubs
Traceback (most recent call last):
  File "train.py", line 76, in <module>
    batch_size=args.batch_size, train_split=args.train_split)
  File "train.py", line 35, in train
    tubgroup = TubGroup(tubs)
  File "/home/pi/.local/lib/python3.5/site-packages/env/lib/python3.5/site-packages/donkeypart_tub/part.py", line 634, in __init__
    self.tubs = [Tub(path) for path in tub_paths]
  File "/home/pi/.local/lib/python3.5/site-packages/env/lib/python3.5/site-packages/donkeypart_tub/part.py", line 634, in <listcomp>
    self.tubs = [Tub(path) for path in tub_paths]
  File "/home/pi/.local/lib/python3.5/site-packages/env/lib/python3.5/site-packages/donkeypart_tub/part.py", line 72, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check your tub path or provide meta info to create a new tub.

Schmill, 2019-07-25:
(env) pi@raspberrypi:~/donkey2 $ python train.py C:\Users\erdoe.LAPTOP-E8A1PPIN\Desktop\tub

This doesn't seem to make sense - from the prompt it looks like you are logged into the Pi, yet you are providing what looks like a Windows style path & location for the tubs??

tawnkramer, 2019-07-25:
And I don’t think you want to train on the pi. Unless you want to wait for
a few days.

python1995, 2019-07-26:
> (env) pi@raspberrypi:~/donkey2 $ python train.py C:\Users\erdoe.LAPTOP-E8A1PPIN\Desktop\tub
> 
> What you have there doesn't seem to make sense - from the prompt it looks like you are logged into the Pi, yet you are providing what looks like a Windows style path & location for the tubs??

Im using MobaXterm  and then  did the test runs and all of the photos are stored in a folder
I did this before and it worked but now Im getting of that error

tawnkramer, 2019-08-06:
please run the train command on your pc.

closed 2019-08-06 by tawnkramer