autorope/donkeycar #442: makemovie salient cropping

ebswift, 2019-07-30:
I don't think this is related to #438 because I get the movie to make to completion without errors with salient.  If I train a model with roi_crop set, then makemovie against this model, the salient highlighting displays over the whole frame, it does not skip the cropped area as expected.  The idea would be to train with cropping, then use salient to look at the results.

Discussion here:
https://donkeycar.slack.com/archives/C4HR56WN6/p1564444240093600

Schmill, 2019-07-30:
Which codebase are you using? (Which version and branch?)

ebswift, 2019-07-30:
3.02 dev revision 6a613f52

tawnkramer, 2019-08-06:
should be fixed now in dev. Will try to roll into master soon.

closed 2019-08-06 by tawnkramer