autorope/donkeycar #448: TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'

dmccreary, 2019-08-18:
Hi folks,

I am using the Donkey 3.1.0 on the Nvidia Nano and I am getting the following error when I use the manage drive function.  I am also testing the Logitech F710 joystick which may or may not be related to this error.

From what I can tell, this appears a type error in some map_range function.

Traceback (most recent call last):
  File "manage.py", line 553, in <module>
    meta=args['--meta'])
  File "manage.py", line 542, in drive
    max_loop_count=cfg.MAX_LOOPS)
  File "/home/dan/ws/donkeycar/donkeycar/vehicle.py", line 144, in start
    self.update_parts()
  File "/home/dan/ws/donkeycar/donkeycar/vehicle.py", line 194, in update_parts
    outputs = p.run(*inputs)
  File "/home/dan/ws/donkeycar/donkeycar/parts/actuator.py", line 181, in run
    self.left_pulse, self.right_pulse)
  File "/home/dan/ws/donkeycar/donkeycar/utils.py", line 219, in map_range
    y = ((x-X_min) / XY_ratio + Y_min) // 1
TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'
GST_ARGUS: 
PowerServiceHwVic::cleanupResources

krissdap, 2019-09-06:
Hi,

I also experienced same issue.
I was using PS4 controller, so the issue is not exclusive to your F710.
The other difference is I'm using Raspbian (2019-04-08-raspbian-stretch-lite), not Ubuntu.

```
  File "/home/pi/projects/donkeycar/donkeycar/utils.py", line 245, in map_range
    y = ((x-X_min) / XY_ratio + Y_min) // 1
TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'
```

shijq23, 2019-09-15:
This can happen if drive without a model "python manage.py drive --js"
when drive mode switches to "local" or "local_angle", then None (angle) can be passed to PWMSteering run function. 

A simple solution is to change class DriveMode to following

```
    class DriveMode:
        def run(self, mode, 
                    user_angle, user_throttle,
                    pilot_angle, pilot_throttle):
            if mode == 'user': 
                return user_angle, user_throttle
            
            elif mode == 'local_angle':
                return pilot_angle if pilot_angle else 0.0, user_throttle
            
            else: 
                return pilot_angle if pilot_angle else 0.0, pilot_throttle * cfg.AI_THROTTLE_MULT if pilot_throttle else 0.0
```

Ezward, 2019-11-27:
@shijq23 I've created a pull request based on the fix you suggested.  https://github.com/autorope/donkeycar/pull/505

closed 2020-02-14 by tawnkramer