autorope/donkeycar #451: Jestson Nano Crashing on Loading Camera

adricl, 2019-08-25:
So I am getting a crash with my Jeson Nano and Pi v2 Camera in Donkeycar 3.1.

The issue is around the capture height and width of the CSI Camera, I think it was to large for the output of my camera or something like that and produced a crash. I fixed it by reducing it to the output height and width.

So my question is the capture height and width supposed to be output height and width? Or is there speed improvements in using the full sensor data? @tawnkramer 

Do you want me to pr a fix for this?

So in the following log I have printed the gstreamer nvidia commands. 

Starting vehicle...
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=3280, height=2464, format=(string)NV12, framerate=(fraction)20/1 ! nvvidconv flip-method=2 ! nvvidconv ! video/x-raw, width=(int)160, height=(int)120, format=(string)BGRx ! videoconvert ! appsink
Device name: Logitech Gamepad F710
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3280 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3280 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 0 
   Output Stream W = 3280 H = 2464 
   seconds to Run    = 0 
   Frame Rate = 21.000000 
GST_ARGUS: PowerService: requested_clock_Hz=35358400
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
NvDdkVicConfigure Failed
nvbuffer_transform Failed
gst_nvvconv_transform: NvBufferTransform Failed 
GST_ARGUS: Cleaning up
GST_ARGUS: 
PowerServiceHwVic::cleanupResources
CONSUMER: Done Success
GST_ARGUS: Done Success

(python:6178): GStreamer-CRITICAL **: 17:29:20.603: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wombat/Documents/donkey_env_3/donkeycar/donkeycar/parts/camera.py", line 174, in update
    self.init_camera()
  File "/home/wombat/Documents/donkey_env_3/donkeycar/donkeycar/parts/camera.py", line 169, in init_camera
    self.poll_camera()
  File "/home/wombat/Documents/donkey_env_3/donkeycar/donkeycar/parts/camera.py", line 181, in poll_camera
    self.frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
cv2.error: OpenCV(4.0.0) /home/wombat/Documents/temp/opencv-4.0.0/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

tawnkramer, 2019-08-25:
The fix I found is to change the output resolution to 224 x 224. I don't
know why the driver crashes when we use the default 160 x 120.

adricl, 2019-08-25:
The master branch has a resolution of width=3280, height=2464. The dev branch has the resolution of 224 x 224. Do you want me to fix this?

tawnkramer, 2019-08-26:
No. Don’t change the capture resolution. Change the output resolution by
adjusting
IMAGE_W = 224
IMAGE_H = 224

In myconfig.py

closed 2019-09-06 by tawnkramer

BrianHenryIE, 2021-08-14:
I was having the same error with Donkey Car 4.2.1: `(Assertion failed) !_src.empty() in function 'cvtColor').`

I was able to use the `nvgstcapture-1.0` tool (as advised on Nvidia's [Taking Your First Picture with CSI or USB Camera](https://developer.nvidia.com/embedded/learn/tutorials/first-picture-csi-usb-camera) page) to troubleshoot that the camera itself was properly connected and working.

Resolution was correct in myconfig.py.

I didn't figure out the cause of the issue. I just flashed my SD card again and it worked immediately then.