autorope/donkeycar #455: No TARGET_H, TARGET_W, TARGET_D

coolerking, 2019-09-04:
I ran as follows:
python manage.py train --model models/foo.h5

but error occurs in train.py line463

```python
                    img_arr = np.array(inputs_img).reshape(batch_size,\
                        cfg.TARGET_H, cfg.TARGET_W, cfg.TARGET_D)
```

because of no defined parameters(TARGET_H, TARGET_W, TARGET_D.

so I wrote them to myconfig.py (or wrote as follows) and ran again without error.

```python
                    img_arr = np.array(inputs_img).reshape(batch_size,\
                        cfg.IMAGE_H, cfg.IMAGE_W, cfg.IMAGE_DEPTH)
```

tawnkramer, 2019-09-06:
seems like your setup is out of sync. try re-installing with latest instructions. Also, please include all output when reporting an error, as well as what platform.

tawnkramer, 2019-09-25:
Please update us if still having this issue with latest.

closed 2019-09-25 by tawnkramer