autorope/donkeycar #457: unable to train on JetsonNano

gbr1, 2019-09-23:
Hi!
I'm trying to train a track on my jetson nano.
After doing some laps I run **manage.py train**
```
(env) gbr1@GBjetracer:~/mycar$ python manage.py train --model ./models/mypilot.h5
using donkey v3.1.0 ...
loading config file: /home/gbr1/mycar/config.py
loading personal config over-rides

config loaded
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
2019-09-23 21:42:43.608971: W tensorflow/core/platform/profile_utils/cpu_utils.cc:98] Failed to find bogomips in /proc/cpuinfo; cannot determine CPU frequency
2019-09-23 21:42:43.610042: I tensorflow/compiler/xla/service/service.cc:161] XLA service 0xe37a1e0 executing computations on platform Host. Devices:
2019-09-23 21:42:43.610101: I tensorflow/compiler/xla/service/service.cc:168]   StreamExecutor device (0): <undefined>, <undefined>
2019-09-23 21:42:43.693323: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:965] ARM64 does not support NUMA - returning NUMA node zero
2019-09-23 21:42:43.693662: I tensorflow/compiler/xla/service/service.cc:161] XLA service 0xbca5650 executing computations on platform CUDA. Devices:
2019-09-23 21:42:43.693714: I tensorflow/compiler/xla/service/service.cc:168]   StreamExecutor device (0): NVIDIA Tegra X1, Compute Capability 5.3
2019-09-23 21:42:43.694055: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1433] Found device 0 with properties: 
name: NVIDIA Tegra X1 major: 5 minor: 3 memoryClockRate(GHz): 0.9216
pciBusID: 0000:00:00.0
totalMemory: 3.87GiB freeMemory: 1.85GiB
2019-09-23 21:42:43.694111: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1512] Adding visible gpu devices: 0
2019-09-23 21:42:44.979774: I tensorflow/core/common_runtime/gpu/gpu_device.cc:984] Device interconnect StreamExecutor with strength 1 edge matrix:
2019-09-23 21:42:44.979847: I tensorflow/core/common_runtime/gpu/gpu_device.cc:990]      0 
2019-09-23 21:42:44.979882: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1003] 0:   N 
2019-09-23 21:42:44.980073: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1115] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 1428 MB memory) -> physical GPU (device: 0, name: NVIDIA Tegra X1, pci bus id: 0000:00:00.0, compute capability: 5.3)
matplotlib not installed
"get_model_by_type" model Type is: linear
WARNING:tensorflow:From /home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/ops/resource_variable_ops.py:435: colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
Instructions for updating:
Colocations handled automatically by placer.
WARNING:tensorflow:From /home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/keras/layers/core.py:143: calling dropout (from tensorflow.python.ops.nn_ops) with keep_prob is deprecated and will be removed in a future version.
Instructions for updating:
Please use `rate` instead of `keep_prob`. Rate should be set to `rate = 1 - keep_prob`.
WARNING:tensorflow:From /home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/keras/utils/losses_utils.py:170: to_float (from tensorflow.python.ops.math_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.cast instead.
training with model type <class 'donkeycar.parts.keras.KerasLinear'>
__________________________________________________________________________________________________
Layer (type)                    Output Shape         Param #     Connected to                     
==================================================================================================
img_in (InputLayer)             (None, 224, 224, 3)  0                                            
__________________________________________________________________________________________________
conv2d_1 (Conv2D)               (None, 110, 110, 24) 1824        img_in[0][0]                     
__________________________________________________________________________________________________
dropout (Dropout)               (None, 110, 110, 24) 0           conv2d_1[0][0]                   
__________________________________________________________________________________________________
conv2d_2 (Conv2D)               (None, 53, 53, 32)   19232       dropout[0][0]                    
__________________________________________________________________________________________________
dropout_1 (Dropout)             (None, 53, 53, 32)   0           conv2d_2[0][0]                   
__________________________________________________________________________________________________
conv2d_3 (Conv2D)               (None, 25, 25, 64)   51264       dropout_1[0][0]                  
__________________________________________________________________________________________________
dropout_2 (Dropout)             (None, 25, 25, 64)   0           conv2d_3[0][0]                   
__________________________________________________________________________________________________
conv2d_4 (Conv2D)               (None, 23, 23, 64)   36928       dropout_2[0][0]                  
__________________________________________________________________________________________________
dropout_3 (Dropout)             (None, 23, 23, 64)   0           conv2d_4[0][0]                   
__________________________________________________________________________________________________
[... 29 lines cut ...]
found 0 pickles writing json records and images in tub /home/gbr1/mycar/data/tub_7_19-09-23
found 0 pickles writing json records and images in tub /home/gbr1/mycar/data/tub_2_19-09-23
/home/gbr1/mycar/data/tub_1_19-09-23
/home/gbr1/mycar/data/tub_3_19-09-23
/home/gbr1/mycar/data/tub_8_19-09-23
/home/gbr1/mycar/data/tub_4_19-09-23
/home/gbr1/mycar/data/tub_6_19-09-23
/home/gbr1/mycar/data/tub_5_19-09-23
/home/gbr1/mycar/data/tub_7_19-09-23
/home/gbr1/mycar/data/tub_2_19-09-23
collating 4124 records ...
train: 3299, val: 825
total records: 4124
steps_per_epoch 25
WARNING:tensorflow:From /home/gbr1/env/lib/python3.6/site-packages/tensorflow/python/ops/math_ops.py:3066: to_int32 (from tensorflow.python.ops.math_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.cast instead.
Epoch 1/100
2019-09-23 21:43:01.964407: I tensorflow/stream_executor/dso_loader.cc:153] successfully opened CUDA library libcublas.so.10.0 locally
Killed
```

As you can see, something go always wrong and killed.

I don't understand what is wrong.

tawnkramer, 2019-09-23:
If you are training on the nano a few tweaks will help. Set CACHE_IMAGES =
False in myconfig.py. Otherwise when it starts to use virtual memory it
will thrash hard and slow or crash. Set BATCH_SIZE = 16 to use less gpu
memory. Booting in run level 3 will free more memory - both gpu and system
ram. And power over barrel connector and put in 10w mode for a 2x boost.

gbr1, 2019-09-23:
@tawnkramer thanks! It works perfectly now!!!

I was already on 10W, just setted BATCH_SIZE=16 and CACHE_IMAGES=False and it works!!!

closed 2019-09-23 by gbr1

RyanC1681, 2019-11-15:
@gbr1 I tried the same thing you did BATCH_SIZE=16 and CACHE_IMAGES=False in 10W, but it still got same error as below:  Any idea? Thanks,
-----
found 0 pickles writing json records and images in tub tub_1_19-11-14
Traceback (most recent call last):
  File "./manage.py", line 589, in <module>
    multi_train(cfg, dirs, model, transfer, model_type, continuous, aug)
  File "/home/jetson/mycar/train.py", line 974, in multi_train
    train_fn(cfg, tub, model, transfer, model_type, continuous, aug)
  File "/home/jetson/mycar/train.py", line 354, in train
    records = gather_records(cfg, tub_names, opts, verbose=True)
  File "/home/jetson/projects/donkeycar/donkeycar/utils.py", line 412, in gather_records
    tubs = gather_tubs(cfg, tub_names)
  File "/home/jetson/projects/donkeycar/donkeycar/utils.py", line 392, in gather_tubs
    tubs = [Tub(p) for p in tub_paths]
  File "/home/jetson/projects/donkeycar/donkeycar/utils.py", line 392, in <listcomp>
    tubs = [Tub(p) for p in tub_paths]
  File "/home/jetson/projects/donkeycar/donkeycar/parts/datastore.py", line 94, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check your tub path or provide meta info to create a new tub.

nvtkaszpir, 2019-11-15:
@RyanC1681 please see the last 2 lines, loks like your files are missing.

gbr1, 2019-11-15:
@RyanC1681 it seems you have any files in your folder.

RyanC1681, 2019-11-15:
Hi, I did have jpg .json in the tub folder for 1 round.  Do I need to record more rounds before train?
Thanks,

RyanC1681, 2019-11-16:
Hi, it finally got trained, but it stuck as below when drive "python manage.py drive --model ~/mycar/models/mypilot.h5" but car didn't move. Any idea? Thanks
CSICamera loaded.. .warning camera

hydernaqvi, 2020-03-12:
I'm not sure if posting here will open the thread.. I'm seeing similar things as what people are seeing above, but my training ends with the following lines:
 

>  File "/usr/lib/python3.5/json/decoder.py", line 339, in decode
>     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
>   File "/usr/lib/python3.5/json/decoder.py", line 357, in raw_decode
>     raise JSONDecodeError("Expecting value", s, err.value) from None
> json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)