autorope/donkeycar #459: gstreamer issues with Nvidia Nano CSI Camera

wolfeidau, 2019-09-28:
The latest changes which were merged in https://github.com/autorope/donkeycar/commit/ee4674f71c81373b98f7ace9e5f6578fc45fe476 result in an opencv error indicating that the camera.get is returning empty frames.

Reverting this change restores my car to a working state.

Anyone else seeing issues with this change?

tawnkramer, 2019-09-28:
> The latest changes which were merged in ee4674f

These changes came from looking at Nvidia’s jetson nano jetcar platform
code. You will need to change the image w and h in myconfig.py to 224 x
224.

wolfeidau, 2019-09-29:
@tawnkramer I ran "createcar" to initialise a new project and this didn't seem to work either, is there a way we can make these requirement more obvious in that `myconfig.py`?

Also would it be helpful to add a note here as well? https://github.com/autorope/donkeycar/blob/dev/docs/guide/create_application.md#camera-setup

macfly1202, 2019-09-29:
I encounter exact same issue. Thanks @tawnkramer for advice H,W : 224,224 !

wolfeidau, 2019-10-07:
I updated `myconfig.py` to use the 224 settings and the are :+1: as below
```
IMAGE_W = 224
IMAGE_H = 224
```

However due to the quantity of data I need to adjust training settings to disable loading all images in memory in `myconfig.py`. I was training with about 12k of images and I exhausted the ram on a laptop with 16GB of RAM. 

This setting fixed it.
```
CACHE_IMAGES = False
```

macfly1202, 2019-10-07:
Following to this change on w,h image, another issue occurs with default model. 

`  File "/home/jmp/Dropbox/github/donkeycar/car/donkeycar/donkeycar/parts/keras.py", line 164, in run
    outputs = self.model.predict(img_arr)
  File "/home/jmp/anaconda3/envs/donkey/lib/python3.7/site-packages/tensorflow/python/keras/engine/training.py", line 1096, in predict
    x, check_steps=True, steps_name='steps', steps=steps)
  File "/home/jmp/anaconda3/envs/donkey/lib/python3.7/site-packages/tensorflow/python/keras/engine/training.py", line 2382, in _standardize_user_data
    exception_prefix='input')
  File "/home/jmp/anaconda3/envs/donkey/lib/python3.7/site-packages/tensorflow/python/keras/engine/training_utils.py", line 362, in standardize_input_data
    ' but got array with shape ' + str(data_shape))
ValueError: Error when checking input: expected img_in to have shape (224, 224, 3) but got array with shape (120, 160, 3)
(donkey) jmp@deep:~/mysim$ ` This is quite easy to change but for the sake of good documentation and good parameter in model for Jetson Nano ...

macfly1202, 2019-10-09:
@tawnkramer yes for sure. It's a minor issue when use Jetson nano in case of copying myconfig to sim folder path.

wolfeidau, 2019-10-20:
@macfly1202 @tawnkramer I get the same error after training and attempting to use the model to drive. Same settings are used used for drive and training, now trying to chase down why this shape error is raised.

macfly1202, 2019-10-20:
@wolfeidau I set the standard setup for myconfig.py use with ```
# #CAMERA
CAMERA_TYPE = "CSIC"   # (PICAM|WEBCAM|CVCAM|CSIC|V4L|MOCK)
IMAGE_W = 160   # deft 160
IMAGE_H = 120    # 120
IMAGE_DEPTH = 3         # default RGB=3, make 1 for mono
CAMERA_FRAMERATE = DRIVE_LOOP_HZ ....DONKEY_GYM = True
DONKEY_SIM_PATH = "/home/jmp/Dropbox/github/donkeycar/simulator/20191007/DonkeySimLinux/donkey_sim.x86_64"
DONKEY_GYM_ENV_NAME = "donkey-generated-track-v0" # ("donkey-generated-track-v0"|"donkey-generated-roads-v0"|"donkey-warehouse-v0"|"donkey-avc-sparkfun-v0")``` on myconfig.py. Do it simple ! Nevertheless I will try to reshape later to use 224*224...

a652311279, 2019-10-31:
Thx sharing
i also has  the problem [**but got array with shape ' + str(data_shape)) ValueError: Error when checking input: expected img_in to have shape (120, 160, 3) but got array with shape (120, 160, 3)** ] 
Since i changed the myconfig.py **image.w** and **image.h**  then i  collect data  and put in the google colab to traing . After trainng, i used the model to drive i got the error. So i need to modify the colab myconfig.py ?

wolfeidau, 2019-11-17:
Just an update, managed to get training and auto pilot running with a small change to the codebase. 

https://github.com/wolfeidau/donkeycar/commit/3b68d578fda8d107ba549d58f2e284768fb56bc9

Need to figure out how to thread these values through from the configuration file.

One thing to note is this also increased the power use of the Jetson Nano so I am unable to use the micro usb connection and have switched to the barrel jack to supply more power. Device literally powers off once it starts inference.

closed 2020-02-14 by tawnkramer