autorope/donkeycar #464: problem when train the donkey at

kiki1112, 2019-10-07:
i have trouble with when i use  

python ~/mycar/manage.py train --model ~/mycar/models/mypilot.h5

and 

rsync -r pi@<your_pi_ip_address>:~/mycar/data/  ~/mycar/data/

when i use train then it run but always displayed failed to load image ,i am confused with that .

Is this possible reason of that i’m not use the usage of rsync to transfer to my computer?

Instead of, i use sd card to copy to my computer.

i didn’t use the rsync because when i use rsync usage ,the terminal shows that usage is wrong and i don’t know why, btw my computer is windows .
Is windows’s usage different with linux?

tawnkramer, 2019-10-09:
Please include some detail about the command you are giving and the output and error.

kiki1112, 2019-10-11:
error is like this
[Errno 2] No such file or di rectory:'C:\\Users\\USER\\mycar\\data\\ tub_4_19-10-06\\13434_cam-image_array_ .j pg failed to load image: C: \Users\USER\mycar\data\ tub_4_19-10-06\13434_cam-image_array_.jpg
 i found that my pi’s python version is 3.5.3 but my pc’s python is 3.7.4  is that cause the problem? i don’t know how to let the version be same, i download miniconda with python3.5.3 version and download donkey again but python always become 3.7.4. please help me(;´༎ຶД༎ຶ`)thanks a lot.

kiki1112, 2019-10-12:
i train the donkey but it shows that ERR>> Unknown extension type on model file!!

tawnkramer, 2019-10-12:
Please follow directions at docs.donkeycar.com

kiki1112, 2019-10-13:
I follow directions then i run my car it did well thank you!

closed 2019-10-13 by kiki1112