autorope/donkeycar #472: How to disable the camera only for controlling the car?

MRinel, 2019-10-24:
Hi, 
I'm new about donkeycar. 
I want to disable the camera bucause it will be occupied when running other programs that need the camera. What I need is to control the car forward or backward, without the assistance of the camera.
Any advice?

Thank you!

dlarue, 2019-10-24:
Donkey does not work without the camera... What data do you intend to send to the NN to train on if not the camera images?

If you just want a remote controlled car, why not look to the Raspberry Pi forums since the purpose of donkey software is to use the camera stream into the CNN to control the vehicle autonomously.

MRinel, 2019-10-24:
@dlarue Thank you for the advice. I will try.
Good luck~

closed 2019-10-24 by MRinel