autorope/donkeycar #501: Downloading Tensorflow work around

wallarug, 2019-11-23:
Hi All,

I had some issues downloading Tensorflow with the instructions on the docs.  It seems pip sometimes plays up when you try to download it.

I have the work around here:

```
wget https://www.piwheels.org/simple/tensorflow/tensorflow-1.13.1-cp37-none-linux_armv7l.whl
pip install tensorflow-1.13.1-cp37-none-linux_armv7l.whl
```

I think it might be worth adding a short note in the Step 11 about failed Tensorflow downloads with pip.

Ezward, 2019-11-26:
Hi @wallarug, I am assuming, but would like to confirm, that this happened while installing donkeycar on the raspberry pi?  What it in [step 11](http://docs.donkeycar.com/guide/robot_sbc/setup_raspberry_pi/#step-11-install-donkeycar-python-code), in this section:
```
git clone https://github.com/autorope/donkeycar
cd donkeycar
git checkout master
pip install -e .[pi]
pip install tensorflow==1.13.1
```

Thanks for reporting this.

wallarug, 2019-11-26:
Hi @Ezward ,

Yes it was this section.  It fails at the last step to install tensorflow.

```
pip install tensorflow==1.13.1
```

This was extensively discussed with doug and others on the #random channel in slack.  It appears to be a pip issue.

I could download it with the wget command above.

dlarue, 2019-11-26:
is there a way to get the file URL from pip in a script or something so that yet another hardcoded URL doesn't have to be added?

wallarug, 2019-11-26:
> is there a way to get the file URL from pip in a script or something so that yet another hardcoded URL doesn't have to be added?

I completely understand doug.   I am not sure.  

It looks like Donkey Car will soon be ditching TF1 anyway once support for TF2 picks up in buster.  So this may only be a temporary workaround for anyone who hits the issue.

Then again - after our discussion on slack, maybe this is more common 🤣

dlarue, 2019-11-26:
I don't think it makes sense to put a 'work around' in the documentation for a temporary pip backend issue.

Ezward, 2019-11-26:
So we need to pin to an explicit version, certainly while we are in TF1 land, because of things that depend upon it;  I can't remember what those things are at this point.  That being said, I believe we can go to TF 1.14 safely.

dlarue, 2019-11-26:
We often pin to a version because we want the instructions to work and let the builder decide if they want to change things once things are working.

As for going to TF 1.14 it might be ok but it has to be explicitly tested with different supported model types and with compatibility with the training generated models.

But I really doubt changing the pip install to get 1.14 instead of 1.13.1 would have changed the temporary error with the pip backend system.

wallarug, 2019-11-27:
What is a temporary issue?  I still cannot do this even 5 days later.

nvtkaszpir, 2019-11-27:
please provide info about:
- python --version
- pip debug

Some suggestions:
- upgrade pip and ensure to use `python3 -m pip` instead of `pip`, such as:
```
python3 -m pip install -e .[pi]
python3 -m pip install tensorflow==1.13.1
```
- set up http proxy in local network (for example squid is quite easy)
- use other mirrors if they exist 
- mayke your own pip mirror with filtering packages only for rpi to limit downloads - https://pypi.org/project/bandersnatch/
- make your own alternative pip repo via https://docs.pulpproject.org/index.html
- try to lower MTU - @wallarug it seems you have issues with your network/internet-service-provider and something weird is going on when you download packages from the pip repo.
For start I suggest lowering MTU on your rpi to 1300 (default is 1500) and see if it helps.
This may be an issue especially that you are from the other part of the globe (Sydney) while rpi packages are from UK.

wallarug, 2020-10-18:
Resolved by DC V4 not requiring the fixed version of TF1

closed 2020-10-18 by wallarug