autorope/donkeycar #511: Can donkeycar drive with trained tub photos?

RyanC1681, 2019-12-13:
Can donkey car drive with trained tub photos by set CAMERA_TYPE = "MOCK" then use tub as input option as below?
i.e. python manage.py drive [--tub=<tub1,tub2,..tubn>] [--model=] ## to import the real tub photo to drive then output pilot/angle', 'pilot/throttle to test if model works.

Thanks,

tawnkramer, 2019-12-13:
Please see usage at
http://docs.donkeycar.com/guide/simulator/

RyanC1681, 2020-01-13:
I couldn't get the simulator work, but I original intend is to import the recorded photo from drive to verify if the trained model work as expected.  Where could I find the code to dump the steer and throttle data during drive?  How to get donkey_path.pkl or train localizer? When I set TRAIN_LOCALIZER = True then it shown KEYError:location during train
Thanks,

Bleyddyn, 2020-05-09:
I'm not sure this is what you mean but there's a tubplot command:

donkey tubplot --help

e.g. donkey tubplot --model=models/Simulator4.h5 --tub=data/20200509_28.tub

Ezward, 2023-03-04:
This can be done in the DonkeyUI in the [Pilot Arena](https://docs.donkeycar.com/utility/ui/#the-pilot-arena)

closed 2023-03-04 by Ezward