autorope/donkeycar #529: donkey create car error

balakreshnan, 2020-01-05:
(env) pi@autocar:~/projects/donkeycar $ donkey createcar --path ~/mycar
using donkey v3.1.1 ...
Traceback (most recent call last):
  File "/home/pi/env/bin/donkey", line 11, in <module>
    load_entry_point('donkeycar', 'console_scripts', 'donkey')()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/projects/donkeycar/donkeycar/management/base.py", line 11, in <module>
    from donkeycar.parts.datastore import Tub
  File "/home/pi/projects/donkeycar/donkeycar/parts/datastore.py", line 16, in <module>
    import pandas as pd
  File "/usr/lib/python3/dist-packages/pandas/__init__.py", line 56, in <module>
    import pandas.util.testing
  File "/usr/lib/python3/dist-packages/pandas/util/testing.py", line 22, in <module>
    from numpy.testing.decorators import slow     # noqa
ImportError: No module named 'numpy.testing.decorators'
(env) pi@autocar:~/projects/donkeycar $

any help is much appreciated.

deltaflyer, 2020-01-08:
Hello @balakreshnan from your prompt, I see that you are not root. Your command `donkey createcar --path /mycar` says that you want to create the folder `car` in the root dir "/". But in Linux only root is allowed to do that. The correct command might be `donkey createcar --path ~/mycar`  . The `~` expands to home directory of the user _pi_ which can be written by the user. Can you re-run the command with `--path ~/mycar`?

balakreshnan, 2020-01-08:
Still the same error ran donkey createcar --path ~/mycar

thanks
Bala

BudBernhard, 2020-01-10:
Hey we took a look at this issue in the donkey car slack #help channel. The issue is that numpy... or pandas however you want to think about it (you need to have numpy for pandas). Anyway it's a version issue. Try pip install numpy==1.16.2 or/and pip install pandas==0.25.3

nvtkaszpir, 2020-01-10:
> Hey we took a look at this issue in the donkey car slack #help channel. The issue is that numpy... or pandas however you want to think about it (you need to have numpy for pandas). Anyway it's a version issue. Try pip install numpy==1.16.2 or/and pip install pandas==0.25.3

this actually requres pandas from the system packages - please upgrade your system first to buster.

BudBernhard, 2020-01-13:
@nvtkaszpir were you replying to me? What version of Numpy and Pandas should we be on if we have buster?

nvtkaszpir, 2020-01-13:
> @nvtkaszpir were you replying to me? What version of Numpy and Pandas should we be on if we have buster?

please read carefully my previous message.

xidianzhuxi, 2020-01-18:
donkey createcar --path ~/mycar
i try to run this on windows10 just show cant find command donkey。whereis the  command donkey

nvtkaszpir, 2020-01-18:
> donkey createcar --path ~/mycar
> i try to run this on windows10 just show cant find command donkey。whereis the command donkey

please do not hijack the thread issue, also it is well documented that under Windows you must run `Anaconda Prompt`

mattpritchard, 2020-02-09:
I ran into the same issue during the createcar step and commented on the Slack channel. At @nvtkaszpir suggestion, I re-installed using Buster and everything was smooth from that point forward. I am using a Pi 3 B+

TCIII, 2020-02-12:
I found that not only did I need to install Buster lite instead of Stretch lite I also had to install pip to be able to get tensorflow and OpenCV to install and import correctly.
I still have a problem installing the correct branch of donkeycar. I am using a ROBO HAT MM1 and require the "dev" branch that has the correct "config.py" for the USE_ROBO HAT MM1.
I have tried using both "git checkout master" and "git checkout dev" and get the same results unless I am doing something wrong. When I use "git checkout dev" it informs me that it has been merged with the master.
Can anyone recommend a solution to this issue? Should I open a separate issue?

tawnkramer, 2020-02-14:
TCIII open a separate issue, or resolve on forums please.

closed 2020-02-14 by tawnkramer