autorope/donkeycar #531: web broswer show nothing

xidianzhuxi, 2020-01-19:
after i run python3 manage.py drive。shell tell me go to the ip:8887 to control the car。
but i cant connet this web page just like this
[image]

nvtkaszpir, 2020-01-19:
Does it show anything more after `Init ESC`  message in the console?

xidianzhuxi, 2020-01-19:
nothing show after `Init ESC`

nvtkaszpir, 2020-01-19:
Then make sure you have followed the procedure in [creating an donkeycar app ](http://docs.donkeycar.com/guide/create_application/) and [calibrating steering](http://docs.donkeycar.com/guide/calibrate/) because it should show something like:
```
...
PWM Throttle created
Adding part PWMSteering.
Adding part PWMThrottle.
Tub does NOT exist. Creating new tub...
New tub created at: /home/pi/mycar/data/tub_2_20-01-19
Adding part TubWriter.
...
...
Starting vehicle at 20 Hz
```
And then it should show web intrface.

xidianzhuxi, 2020-01-19:
thanks,i fix it .one year ago,i build donkeycar on my own board and modify the actuator.py。i noticed that the new version seems to support imu?how can i use it?

xidianzhuxi, 2020-01-19:
sorry but a new problem i find that the web page i can not control.no use to press the button in red zone shows below .and i can not click to drag neither
[image]

nvtkaszpir, 2020-01-19:
On the left in `Mode & Pilot` select mode `User`.
On the top left `Control Mode` select `Joystick`.

Then you should be able to use other controls.

Also ensure not to have any special adblockers or noscript addons enabled.

xidianzhuxi, 2020-01-19:
I can select mode User in `Mode & Pilot` but i can not select `Joystick` on the top left `Control Mode`.
I press the key space and go to see the tub,no image and json file stored.F12 shows 4 js error.
[image]

xidianzhuxi, 2020-01-19:
by the way: i run `python3 manage.py drive --js`  all is well

closed 2020-01-25 by tawnkramer

CarloHan, 2020-06-09:
Hi, I have the same problem with you, the website shows nothing. May I ask how did you solve the problem? Thx!