autorope/donkeycar #536: Can donkeycar support 2x cameras?

RyanC1681, 2020-01-30:
Can donkeycar support 2x cameras?
In the line as below shown 2x cam-image_arrary_a/b, but only 1 output cam-image_array.
Could it output 2x camera arrary, If I hook up 2x CSI or Webcam?  
Can CSI mix with USB WebCAM?
-----------------
V.add(StereoPair(), inputs=['cam/image_array_a', 'cam/image_array_b'], 
            outputs=['cam/image_array'])  => outputs=['cam/image_array_a',  'cam/image_array_b']
            # can it output 2x image files?
Thanks,

tawnkramer, 2020-02-14:
Yes, two camera support can work. Only tested with two usb webcameras. But there's no limit in the software. It can use whatever cameras the hardware supports.

closed 2020-02-14 by tawnkramer

RyanC1681, 2020-02-19:
How to mod camera.py: def gstreamer_pipeline to support 2x CSI cameras?
Thanks,