autorope/donkeycar #545: Camera feed not showing in browser

NicholaiStaalung, 2020-02-19:
Hello,

Trying to set up the donkey car but the camera feed is not showing
[image]

Specs:
Jetson nano with latest jetpack
I **did not** install in a virtual environment on the nano

I already troubleshooted some
ls -l /dev/video0 shows the camera

I tried https://medium.com/@dmccreary/getting-your-camera-working-on-the-nvida-nano-336b9ecfed3a and got a perfect feed

I inspected the page and thought maybe the feed came from a wrong source so i changed the source to /dev/video0 without luck

[image]

Im getting the following traceback in the terminal when im running python3 manage.py drive

> GST_ARGUS: Done Success
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/donkeycph/projects/donkeycar/donkeycar/parts/camera.py", line 172, in update
    self.init_camera()
  File "/home/donkeycph/projects/donkeycar/donkeycar/parts/camera.py", line 167, in init_camera
    self.poll_camera()
  File "/home/donkeycph/projects/donkeycar/donkeycar/parts/camera.py", line 179, in poll_camera
    self.frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
cv2.error: OpenCV(4.1.0) /home/donkeycph/projects/cv2/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor

NicholaiStaalung, 2020-02-19:
This solved it 
https://custom-build-robots.com/raspberry-pi-robot-cars/autonomously-driving-nvidia-jetson-nano-ai-robot-car-configuration-of-the-donkey-car-framework/12137?lang=en

closed 2020-02-19 by NicholaiStaalung

RossMelbourne, 2020-09-07:
I am having the same issue that this issue #545 has solved.  I tried the article from above which had a solution for this exact problem.  However the code changes mentioned do not work for me.

I also tried https://medium.com/@dmccreary/getting-your-camera-working-on-the-nvida-nano-336b9ecfed3a and got a perfect feed.

I am using the latest donkey v3.1.5.

I am using the recommended camera for the Jetson Nano which has the IMX219 Sensor.

Here is myconfig.py entries:
myconfig.com:
CAMERA_TYPE = "CSIC"   
IMAGE_W = 224
IMAGE_H = 224

Here is the error:
Opening /dev/input/js0...
Device name: Nimbus
Starting vehicle at 20 Hz
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp, line 9716
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/rossm/robocar/donkeycar/donkeycar/parts/camera.py", line 174, in update
    self.init_camera()
  File "/home/rossm/robocar/donkeycar/donkeycar/parts/camera.py", line 169, in init_camera
    self.poll_camera()
  File "/home/rossm/robocar/donkeycar/donkeycar/parts/camera.py", line 181, in poll_camera
    self.frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
cv2.error: /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp:9716: error: (-215) scn == 3 || scn == 4 in function cvtColor

I also tried varying the myconfig.py to the default resolution and/or making the resolutions in camera.py match myconfig.py but no joy.

RossMelbourne, 2020-09-11:
I solved this problem by using the instructions here:  https://diyrobocars.com/installation-steps-for-donkeycar-on-jetson-nano-xavier/

I suggest updating the Donkeycar Docs to reflect this.

huntlin, 2022-05-23:
can you tell me how reslove camera problem?