autorope/donkeycar #554: Can't Train DonkeyCar Using Website Instructions

hydernaqvi, 2020-03-11:
I'm at the top of the page at
https://docs.donkeycar.com/guide/train_autopilot/#train-a-model

I'm at the first part where I have to type in :
`rsync -rv --show-progress --partial pi@<your_pi_ip_address>:~/mycar/data/  ~/mycar/data/`

I keep getting the following response:

> rsync: --show-progress: unknown option
> rsync error: syntax or usage error (code 1) at main.c(1585) [client=3.1.2]

I'm going to try to to go ahead and remove the show progress part but it would be good to know why this isn't working

nvtkaszpir, 2020-03-11:
Please provide output of the command
```shell
rsync --version
```

hydernaqvi, 2020-03-11:
rsync  version 3.1.2  protocol version 31
Copyright (C) 1996-2015 by Andrew Tridgell, Wayne Davison, and others.
Web site: http://rsync.samba.org/
Capabilities:
    64-bit files, 64-bit inums, 32-bit timestamps, 64-bit long ints,
    socketpairs, hardlinks, symlinks, IPv6, batchfiles, inplace,
    append, ACLs, xattrs, iconv, symtimes, prealloc

rsync comes with ABSOLUTELY NO WARRANTY.  This is free software, and you
are welcome to redistribute it under certain conditions.  See the GNU
General Public Licence for details.

hydernaqvi, 2020-03-11:
This is the error I'm seeing now
> Unexpected remote arg: @192.168.137.141:~/mycar/data/
> rsync error: syntax or usage error (code 1) at main.c(1361) [sender=3.1.2]
I've been able to access the server using this ip address and the port 8887 on the browser so I don't think the ip address is the issue

nvtkaszpir, 2020-03-11:
empty username?

hydernaqvi, 2020-03-11:
I'm not sure what you mean. Apart from the IP address there's no suggestion I need to change anything else. All other settings are default

nvtkaszpir, 2020-03-11:
OK, indeed my question was a bit vague ;)
```
Unexpected remote arg: @192.168.137.141:~/mycar/data/
```
seems like you are missing username `pi` in the rsync command, such as:
```shell
rsync -rv --show-progress --partial @192.168.137.141:~/mycar/data/ ~/mycar/data/
```
versus
```shell
rsync -rv --show-progress --partial pi@192.168.137.141:~/mycar/data/ ~/mycar/data/
```

hydernaqvi, 2020-03-12:
Thank you so far I've made it past my earlier issues, but I've hit another bump. I'm not sure if I should start a new thread. When I run the next command to run all tubs for training 
 using this command:
`python ~/mycar/manage.py train --model ~/mycar/models/mypilot.h5`

I get the following error towards the end:

> File "/usr/lib/python3.5/json/decoder.py", line 339, in decode
> obj, end = self.raw_decode(s, idx=_w(s, 0).end())
> File "/usr/lib/python3.5/json/decoder.py", line 357, in raw_decode
> raise JSONDecodeError("Expecting value", s, err.value) from None
> json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

hydernaqvi, 2020-03-12:
I'm not sure if it helps but I'm also seeing a lot of lines saying 0 pickles found

tawnkramer, 2020-03-12:
it looks like you might have a corrupt or zero length json file. I've seen
this happen from windows users mainly. Not sure what it up with the rsync
on that platform. But we may want to switch to an FTP based soln at some
point.

In the meantime, you can run the command 'donkey tubcheck ~/mycar/data/*
--fix' and see if that removes the problem records.

https://docs.donkeycar.com/utility/donkey/#check-tub

hydernaqvi, 2020-03-12:
I ran this as suggested:
`donkey tubcheck ~/mycar/data/*
--fix`

I got the following results with errors:

> using donkey v3.1.1 ...
> Traceback (most recent call last):
>   File "/home/pi/env/bin/donkey", line 11, in <module>
>     load_entry_point('donkeycar', 'console_scripts', 'donkey')()
>   File "/home/pi/projects/donkeycar/donkeycar/management/base.py", line 553, in                                                                                                              execute_from_command_line
>     c.run(args[2:])
>   File "/home/pi/projects/donkeycar/donkeycar/management/base.py", line 273, in                                                                                                              run
>     self.check(args.tubs, args.fix, args.delete_empty)
>   File "/home/pi/projects/donkeycar/donkeycar/management/base.py", line 262, in                                                                                                              check
>     tubs = [Tub(path) for path in tub_paths]
>   File "/home/pi/projects/donkeycar/donkeycar/management/base.py", line 262, in                                                                                                              <listcomp>
>     tubs = [Tub(path) for path in tub_paths]
>   File "/home/pi/projects/donkeycar/donkeycar/parts/datastore.py", line 52, in _                                                                                                             _init__
>     self.meta = json.load(f)
>   File "/usr/lib/python3.5/json/__init__.py", line 268, in load
>     parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
>   File "/usr/lib/python3.5/json/__init__.py", line 319, in loads
>     return _default_decoder.decode(s)
>   File "/usr/lib/python3.5/json/decoder.py", line 339, in decode
>     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
>   File "/usr/lib/python3.5/json/decoder.py", line 357, in raw_decode
>     raise JSONDecodeError("Expecting value", s, err.value) from None
> json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

hydernaqvi, 2020-03-13:
To try something different I tried to run one tub for training by using the donkeycar instructions and simply placing the name of the tub folder into the suggested area. It ran but eventually gave an error that the path provided doesn't exist. 

Then I navigated into the data folder and tried the same command again:
`(env) pi@raspberrypi:~/mycar/data $ python ~/mycar/manage.py train --tub tub_10_20-01-01 --model ./models/mypilot.h5
`
This ran for a bit again but eventually resulted in the following error

> Use tf.cast instead.
> Epoch 1/100
> 2020-03-13 03:24:26.556491: W tensorflow/core/framework/allocator.cc:124] Allocation of 55590912 exceeds 10% of system memory.
> terminate called after throwing an instance of 'std::bad_alloc'
>   what():  std::bad_alloc
> Aborted

I'm really not sure what to do at this point and I could really use some advice

Side note: How can I remove all tubs and try again? Would that help?

nvtkaszpir, 2020-03-14:
Please don't run training on rpi, it has not enough memory to do it.
It's even stated in the error message you provided.

hydernaqvi, 2020-03-31:
@nvtkaszpir I was just trying all combinations I could to try and get it to work. I have a partner who worked on this using a Mac with different data and it worked. It could be a compatibility thing. I had hit the update command on my DonkeyCar and I wonder if I need a solution equivalent to replacing  `--show-progress` with `--progress`

I'll most likely delete everything and try starting from scratch in case I changed something that wasn't meant to be changed

PR #555 (2020-04-22): Update train_autopilot.md rsync command
Fixes #554
commit a45ce2c7 (2020-04-22): Update train_autopilot.md

Fixes #554
closed 2020-04-23 by tawnkramer