autorope/donkeycar #557: Install on Jetson Nano requires 3 versions of pip

Ezward, 2020-03-14:
I've recently use the master docs to install on an Nvidia Jetson Nano and I had a bunch of troubles.  https://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/

- The initial install will install pip 20.x.x.  This version apparently has a bug in a virtualenv create with  --system-site-packages.  So you get an error on this step `pip install -e .[nano]`  https://github.com/pypa/pip/issues/6264
- After hitting the bug, I installed pip 19.3.1.  That got farther, but failed later in the  `pip install -e .[nano]`
- I finally installed pip 18.1 and that completed correctly

I noticed in that pip bug that two active donkeycar users, naisy and dmccreary, both ran into the issue and both fixed it by installing pip 19.0

tawnkramer, 2020-03-14:
>    error on this step pip install -e .[nano] pypa/pip#6264

I didn’t have any problems like that. I did try the latest jet pack. Please
feel free to edit docs to apply notes that were helpful. Thanks.

Ezward, 2020-03-25:
I just reinstalled and found that pip 19.1 works without error.

closed 2020-06-04 by tawnkramer