autorope/donkeycar #558: xbox controller  and instructions don't work for Jetson Nano

Ezward, 2020-03-14:
I recently installed donkey on a Jetson Nano and had a lot of problems with the 'xbox' controller.  Note that I did not install any driver; I'm using whatever is installed by default on the Nano. The instructions for getting the xbox controller connected work for the RPi, but not the Nano.  In particular:
- editing `/etc/modprobe.d/xbox_bt.conf` does not seem to work in the Nano.  If you do that and restart, `cat /sys/module/bluetooth/parameters/disable_ertm` continues to always return 'N', so the setting is never used.  I found this [resource](https://www.roboticsbuildlog.com/hardware-control/xbox-one-controller-with-nvidia-jetson-nano) that fixed the issue disable_ertm issue.
- I was not able to get the bluetoothctl connection to work; instead directly connected and used the GUI, which worked after fixing the first issue.  Note that I am NOT saying the bluetoothctl method does not work; we should just test it again after using the resource below to fix the disable_ertm issue.  But it might be easier to suggest just using the Ubuntu GUI if you are directly connected to the Nano and only use the bluetoothctl process is you are in headless mode.
- The mapping of the joystick axis and trigger axis is different on the Nano than the RPi, so the 'xbox' controller type just does not work properly on the Nano.  I had to solve this problem by using the 'donkey createjs` command, which is a totally awesome command, but it also non-obvious.  I've created a separate pull request that makes using the output of the `donkey createjs` command easier, but that does not fix the issue of the "CONTROLLER_TYPE='xbox'" being broken on the nano.

nvtkaszpir, 2020-03-14:
1. can you share output of `lsb_release -a` (may require `apt-get install -y lsb-release`)
and `uname -a`
2. I've noticed that sometimes adding device via bluetoothctl needs few tries (this varies from bt devices, scanning frequency and signal quality). Without disabled ERTM it always fails anyway.
Moreover gui takes some resources (cpu/mem/gpu, not to mention extra system services), but can be disabled after setting up devices.
3. nice :)

GrantMoe, 2020-03-24:
1. I encountered the same problem, and fixed it the same way.
2. The bluetoothctl method worked for me, but only after I nuked and paved to clear away all the muddling I did trying to fix the disable_ertm issue (before I found the roboticsbuildlog article).
3. CONTROLLER_TYPE='xbox' worked fine for me (perhaps this has been fixed since this issue was first opened)

Ezward, 2023-03-04:
docs have been update https://docs.donkeycar.com/parts/controllers/#jetson-nano

closed 2023-03-04 by Ezward