autorope/donkeycar #567: cant run PiCamera

9018, 2020-03-24:
(env) root@kali:~/projects/donkeycar# python /root/mycar/manage.py drive
using donkey v3.1.1 ...
loading config file: /root/mycar/config.py
loading personal config over-rides

config loaded
cfg.CAMERA_TYPE PICAM
cfg.CAMERA_TYPE PICAM
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "/root/mycar/manage.py", line 572, in <module>
    meta=args['--meta'])
  File "/root/mycar/manage.py", line 100, in drive
    cam = PiCamera(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH)
  File "/root/projects/donkeycar/donkeycar/parts/camera.py", line 20, in __init__
    self.camera = PiCamera() #PiCamera gets resolution (height, width)
  File "/root/projects/donkeycar/picamera/picamera/camera.py", line 407, in __init__
    self._init_revision(options)
  File "/root/projects/donkeycar/picamera/picamera/camera.py", line 479, in _init_revision
    with mo.MMALCameraInfo() as camera_info:
  File "/root/projects/donkeycar/picamera/picamera/mmalobj.py", line 2425, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/root/projects/donkeycar/picamera/picamera/mmalobj.py", line 698, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/root/projects/donkeycar/picamera/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented

9018, 2020-03-24:
i use the kaliarm64 on pi4
i use the https://github.com/raspberrypi/userland
i can run vcgencmd get_camera ； supported=1 detected=1
i can use raspistill -o image.jpg -rot 180

adeeb10abbas, 2020-05-06:
I struggled with the same problem for hours before the forehead-slap-worthy realization that I had the camera cable inserted upside down into the raspberry pi's slot :/ Thought I'd post that here in case it helps others :)

DocGarbanzo, 2022-10-27:
Closed due to inactivity

closed 2022-10-27 by DocGarbanzo