autorope/donkeycar #586: Jetson Nano (card-image-r32.4.2) and Tensorflow 1.13

filmo, 2020-05-25:
Following the [nano setup instructions](https://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/) no longer seems to work with current links. In particular, the current (a/o 2020-05-25) SD card nv-jetson-nano-sd-card-image-r32.4.2 comes with Cuda 10.2 on it which is not compatible with the in the subsequent install of TF in the instructions.

https://developer.download.nvidia.com/compute/redist/jp/v42 tensorflow-gpu==1.13.1+nv19.3

Question: what is the correct archival SD card that is compatible with TF 1.13 ?

Alternatively, which version of TF is compatible with the current Cuda 10.2 and ~also~ works with the Donkey Car code base? (For example, could we use the TF 2.x family ? How about TF 1.15 instead of 1.13 ??)

tikurahul, 2020-05-25:
Nvidia updates it images every couple of months. 
https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html is the best place for information about how to install Tensorflow on Jetson. 

Specifically the step you are looking for is:

```
pip install --pre --extra-index-url https://developer.download.nvidia.com/compute/redist/jp/v44 ‘tensorflow<2’
```

This should install TF 1.15.2 on the Nano. You should also refer to the above URL for the pre-requisites.

tawnkramer, 2020-06-04:
@tikurahul do you want to change the docs?

tikurahul, 2020-06-04:
I also need to fix `keras.py`. Will send you a PR today.

NathKorat, 2022-08-09:
you saved my life!!!!

Ezward, 2023-03-04:
Closing as obsolete.

closed 2023-03-04 by Ezward