autorope/donkeycar #589: ConnectionRefusedError: [Errno 111] Connection refused

Saytas, 2020-05-30:
Hello,

I am trying to start the DonkeyGym environment, but I am getting the following error when I run python manage.py drive.

`Traceback (most recent call last):
  File "/home/sytasch/ROS/autonomous_car/auto/project/gym-donkeycar/gym_donkeycar/core/client.py", line 42, in connect
    self.s.connect((self.host, self.port))
ConnectionRefusedError: [Errno 111] Connection refused
   .......   
Exception: Could not connect to server. Is it running? If you specified 'remote', then you must start it manually.
closing donkey sim subprocess
`

[image]

How can I connect to the server? I would appreciate any suggestions. Thanks in advance.

tawnkramer, 2020-06-04:
did you double click on it to start it before running the python manage drive?

Saytas, 2020-06-07:
Thank you for the response. Do you mean the simulation executable? If so, no I did not.

peterstratton, 2020-06-22:
I have the same error, what am I supposed to double click on before running 'python manage.py drive'?

tawnkramer, 2020-06-22:
The simulator executable.

tawnkramer, 2020-06-22:
https://youtu.be/wqQMmHVT8qw

peterstratton, 2020-06-22:
Oh it turns out I wasn't using the newest simulation, I switched to the newest one and it works great, thanks!

closed 2020-06-22 by tawnkramer

MajesticKhan, 2020-07-11:
Hey,

I'm still facing the same issue.

@peterstratton, what did you do in particular after you got the newest version?

@tawnkramer Is the simulator suppose to be running while the manage.py script runs?

Thanks

likefrankie, 2020-07-11:
I am having the same issue. Please see error message below. The simulator starts as expected but the connection is refused. 

```
(donkeyenv) username@system76-pc:~/mysim$ python manage.py drive
using donkey v3.1.2 ...
loading config file: /home/username/mysim/config.py
myconfig myconfig.py
loading personal config over-rides from myconfig.py

config loaded
cfg.CAMERA_TYPE PICAM
starting DonkeyGym env
donkey subprocess started
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
connecting to 127.0.0.1 9091
Traceback (most recent call last):
  File "/home/username/gym-donkeycar/gym_donkeycar/core/client.py", line 42, in connect
    self.s.connect((self.host, self.port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "manage.py", line 644, in <module>
    meta=args['--meta'])
  File "manage.py", line 109, in drive
    cam = DonkeyGymEnv(cfg.DONKEY_SIM_PATH, host=cfg.SIM_HOST, env_name=cfg.DONKEY_GYM_ENV_NAME, conf=cfg.GYM_CONF, delay=cfg.SIM_ARTIFICIAL_LATENCY)
  File "/home/username/donkeycar/donkeycar/parts/dgym.py", line 25, in __init__
    self.env = gym.make(env_name, exe_path=sim_path, host=host, port=port)
  File "/home/username/donkeyenv/lib/python3.7/site-packages/gym/envs/registration.py", line 145, in make
    return registry.make(id, **kwargs)
  File "/home/username/donkeyenv/lib/python3.7/site-packages/gym/envs/registration.py", line 90, in make
    env = spec.make(**kwargs)
  File "/home/username/donkeyenv/lib/python3.7/site-packages/gym/envs/registration.py", line 60, in make
    env = cls(**_kwargs)
  File "/home/username/gym-donkeycar/gym_donkeycar/envs/donkey_env.py", line 136, in __init__
    super(GeneratedTrackEnv, self).__init__(level=3, *args, **kwargs)
  File "/home/username/gym-donkeycar/gym_donkeycar/envs/donkey_env.py", line 49, in __init__
    self.viewer = DonkeyUnitySimContoller(level=level, host=host, port=port, cam_resolution=cam_resolution)
  File "/home/username/gym-donkeycar/gym_donkeycar/envs/donkey_sim.py", line 39, in __init__
    self.client = SimClient(self.address, self.handler)
  File "/home/username/gym-donkeycar/gym_donkeycar/core/sim_client.py", line 25, in __init__
    super().__init__(*address)
  File "/home/username/gym-donkeycar/gym_donkeycar/core/client.py", line 33, in __init__
    self.connect()
  File "/home/username/gym-donkeycar/gym_donkeycar/core/client.py", line 44, in connect
    raise( Exception("Could not connect to server. Is it running? If you specified 'remote', then you must start it manually."))
Exception: Could not connect to server. Is it running? If you specified 'remote', then you must start it manually.
closing donkey sim subprocess

```

tawnkramer, 2020-07-12:
Try stating the sim first.

Here’s a walk through:

https://youtu.be/wqQMmHVT8qw

likefrankie, 2020-07-13:
Tawn,Thank you for your prompt response. I reviewed the videos and followed the instructions, but the issue persist.Please let me know if you have any suggestions. Regards,Francisco

https://youtu.be/wqQMmHVT8qw

likefrankie, 2020-07-15:
Tawn,I am writing to share some good news. I was able to fix the issue. My problem is that I wanted to avoid the minianaconda installation because it could create issues with my other python installations.So I tried to install all the necessary packages on my own. And I think 'opencv' was not properly installed (even though no error messages ocurred).I went through the long binary installation in the link below. And now the connection issue resolved!https://www.pyimagesearch.com/2018/05/28/ubuntu-18-04-how-to-install-opencv/I thought I would share that in case others run into similar installation issues.Thanks again for your response earlier.Regards,FranciscoOn Jul 12, 2020 8:13 PM, francisco@franciscoazeredo.com wrote:Tawn,Thank you for your prompt response. I reviewed the videos and followed the instructions, but the issue persist.Please let me know if you have any suggestions. Regards,Francisco

https://youtu.be/wqQMmHVT8qw