autorope/donkeycar #612: Move Jetson camera.py towards jetcam library

wallarug, 2020-08-12:
Hey Team,

I was just told about this library that apparently handles all camera operations for the Jetson Nano.  It might be worth moving towards this instead of the current raw gstreamer functions.

https://github.com/NVIDIA-AI-IOT/jetcam

TCIII, 2020-08-31:
wallarug,
Presently the Nano has no way to handle the new Pi 12 Mp HQ Camera, though there are some articles on the web that purport to have solved this issue with the Nano.
I suspect that jetcam is being used in the jetson-inference app as seen here: https://github.com/dusty-nv/jetson-inference
Regards,
TCIII

DocGarbanzo, 2021-01-16:
@wallarug - I don't have a nano so cannot test. Do you think you could test that and do a PR, if it works better than what we have?

sctse999, 2021-01-18:
Just quickly checked Jetcam ...  how do we benefit if we move to use it? More external dependencies might bring more problems :)

IMHO, we can still keep the current CSIC even if we introduce JetCam.

wallarug, 2021-01-19:
I'd say close this off.  I agree with Jonathan on this one.

closed 2021-01-19 by wallarug

DocGarbanzo, 2021-01-22:
@wallarug @sctse999 - thank you guys!