autorope/donkeycar #614: Bug: Syntax error in web.py; f-string is not compatible with python 3.5

cfox570, 2020-08-19:
Standard install on raspberry pi 3b+ uses python 3.5.   The code in web.py line 224 uses an f-string which requires python 3.6+.

print(f"wsCalibrate {message}"). replace with print("wsCalibrate ",message)

sctse999, 2020-08-19:
Hello,

Sorry it's my bad to use the f-string. 2 ways to solve this issue. 

1) Update to buster. By default buster should be using python 3.7. I believe buster is working on pi3b+ too. 
2) Submit a PR to fix this. 

I am not sure but I recall that we want to move towards python 3.7. If we are targeting the build to 3.7, then it might be a good idea to move to using buster.

cfox570, 2020-08-19:
What is the performance hit after upgrading to Buster?

sctse999, 2020-08-20:
I didn't test but I think pi3b+ would run buster fine.

DocGarbanzo, 2020-09-06:
The donkey code definitely runs faster on buster on a RPi 3B+. And python 3.5 is not really supported any longer.

closed 2020-11-17 by sctse999