autorope/donkeycar #615: Move away from DRIVE_TRAIN_TYPE and HAVE_

wallarug, 2020-08-20:
Can we please pick one standard?   

Have both is causing some issues for people as it is counterproductive to have to turn something on twice.

My recommendation would be to use only two variables throughout and remove all the `HAVE_BOARD` ones.

`CONTROLLER_TYPE=""`
 `DRIVE_TRAIN_TYPE=""`

tawnkramer, 2020-08-20:
the have sombrero doesn't change the drive train type

DocGarbanzo, 2021-01-16:
@wallarug - is this still an active problem?

zlite, 2021-04-20:
Yes, this is still suboptimal. I agree with @wallarug that we should define inputs (controller type) and outputs (drive trains) and that no other board definition should be required. In this case "HAVE_ROBOHAT" is redundant and should be removed in the code and replaced with either "controller_type" or "drive_train_type" as appropriate

zlite, 2021-04-25:
Fixed with this PR: https://github.com/autorope/donkeycar/pull/851

closed 2021-04-25 by zlite