autorope/donkeycar #620: Bug in hello world example?

cdeil, 2020-08-23:
The example at http://docs.donkeycar.com/#hello-world has a line
```
V.add(tub, inputs=inputs)
```
but no variable `inputs` has been defined.

This will give a NameError, no?

Current code version is [here](https://github.com/autorope/donkeycar/blob/c983d7190a06bf852d46ac3356af703e03e02103/docs/index.md#hello-world).

sctse999, 2020-08-24:
Nice catch. Probably that won't run. Do you want to submit a PR for that?

cdeil, 2020-08-24:
I just got my car yesterday and haven't even assembled it yet, I don't know what the fix is here. @sctse999 - can you please do it?

tawnkramer, 2020-09-12:
looks fixed now

closed 2020-09-12 by tawnkramer

cdeil, 2020-09-13:
@tawnkramer - Please re-open this issue.

The example at https://github.com/autorope/donkeycar/blob/master/docs/index.md#hello-world that gets displayed at http://docs.donkeycar.com/#hello-world is unchanged and still broken.

It's currently this:
```
import donkey as dk

#initialize the vehicle
V = dk.Vehicle()

#add a camera part
cam = dk.parts.PiCamera()
V.add(cam, outputs=['image'], threaded=True)

#add tub part to record images
tub = dk.parts.Tub(path='~/mycar/data',
                   inputs=['image'],
                   types=['image_array'])
V.add(tub, inputs=inputs)

#start the vehicle's drive loop
V.start(max_loop_count=100)
```

Already the first import is broken, it's "donkeycar", not "donkey". Other things about this script are broken, too.

There is a working example using CVCam here: https://github.com/autorope/donkeycar#modify-your-cars-behavior

```
from donkeycar import Vehicle
from donkeycar.parts.camera import PiCamera
from donkeycar.parts.datastore import TubWriter

# initialize the vehicle
V = Vehicle()

IMAGE_W = 160
IMAGE_H = 120
IMAGE_DEPTH = 3

# add a camera part
cam = PiCamera(image_w=IMAGE_W, image_h=IMAGE_H, image_d=IMAGE_DEPTH)
V.add(cam, outputs=['image'], threaded=True)

#while cam.run() is None:
#    time.sleep(1)

# add tub writer part to record images
tub = TubWriter(path='./dat',
                inputs=['image'],
                types=['image_array'])
V.add(tub, inputs=['image'], outputs=['num_records'])

# start the vehicle's drive loop
V.start(rate_hz=10)
```

Following that example I tried this to make PiCamera work:
```
from donkeycar import Vehicle
from donkeycar.parts.camera import PiCamera
from donkeycar.parts.datastore import TubWriter

# initialize the vehicle
V = Vehicle()

IMAGE_W = 160
IMAGE_H = 120
IMAGE_DEPTH = 3

# add a camera part
cam = PiCamera(image_w=IMAGE_W, image_h=IMAGE_H, image_d=IMAGE_DEPTH)
V.add(cam, outputs=['image'], threaded=True)

#while cam.run() is None:
#    time.sleep(1)

# add tub writer part to record images
tub = TubWriter(path='./dat',
                inputs=['image'],
                types=['image_array'])
V.add(tub, inputs=['image'], outputs=['num_records'])

# start the vehicle's drive loop
V.start(rate_hz=10)
```
But it fails like this:
```
$ python b.py
using donkey v3.1.5 ...
PiCamera loaded.. .warming camera
Adding part PiCamera.
/usr/lib/python3/dist-packages/picamera/encoders.py:544: PiCameraResolutionRounded: frame size rounded up from 160x120 to 160x128
  width, height, fwidth, fheight)))
Tub does NOT exist. Creating new tub...
New tub created at: ./dat
Adding part TubWriter.
Starting vehicle at 10 Hz
Traceback (most recent call last):
  File "/home/pi/projects/donkeycar/donkeycar/vehicle.py", line 151, in start
    self.update_parts()
  File "/home/pi/projects/donkeycar/donkeycar/vehicle.py", line 199, in update_parts
    outputs = p.run(*inputs)
  File "/home/pi/projects/donkeycar/donkeycar/parts/datastore.py", line 447, in run
    self.put_record(record)
  File "/home/pi/projects/donkeycar/donkeycar/parts/datastore.py", line 228, in put_record
    img = Image.fromarray(np.uint8(val))
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Shutting down vehicle and its parts...
Stopping PiCamera
Part Profile Summary: (times in ms)
+------+-----+-----+-----+-----+-----+-----+-------+
| part | max | min | avg | 50% | 90% | 99% | 99.9% |
+------+-----+-----+-----+-----+-----+-----+-------+
+------+-----+-----+-----+-----+-----+-----+-------+
```

@tawnkramer , @sctse999 or anyone - do you know how to write a working example with PiCamera?

Otherwise maybe removing the broken example and just pointing to the working one using CVCam in the README could be an option?