autorope/donkeycar #623: Why does GYM_CONF have to be enabled when using the Robo HAT MM1

TCIII, 2020-08-29:
For those of you using the Robo HAT MM1 with either your Rpi 3B+ or 4B, I found that I had to enable GYM_CONF to get either manage.py drive or manage.py drive --js to run without an error. 
I didn't have to enable GYM_CONF when I was running my Rpi 3B+ with an older version of DC 3, but had to with a fresh install of the latest DC 3.1.3 on my Rpi 4B. 
Comments?

DrewTurner, 2020-08-29:
Running into this as well.  DC 3.1.5 using ps4 controller.
`NameError: name 'GYM_CONF' is not defined
`

sctse999, 2020-09-09:
Will look into this

wallarug, 2020-09-10:
Hey @TCIII ,

What/where do you end up putting the variable `GYM_CONF` to make Donkey Car run successfully?  I had a quick look at this one and can only see one reference to `GYM_CONF` in the current manage.py file.

If you can provide the above info, I might be able to patch it.

TCIII, 2020-09-10:
@wallarug,
All I did was enable GYM_CONF and nothing else.
The manage.py drive start up ERROR then went away.

wallarug, 2020-09-10:
Can you check what is in config.py?

TCIII, 2020-09-10:
@wallarug,
Yes it appears to be enabled in config.py, but as I said before I had to enable it in myconfig.py to keep from getting a GYM_CONF ERROR when I ran either manage.py drive or manage.py drive --JS. I do not think that the Robo HAT MM1 software is causing the issue.

tawnkramer, 2020-09-12:
that should be gone now. There was a need to have the guid put in the myconfig for users. That's removed now.

TCIII, 2020-09-12:
@tawnkramer,
I just updated my DC 3.1.5 and disabled GYM_CONF in myconfig.py, ran manage.py drive and it still throws this error:

loading personal config over-rides from myconfig.py
Traceback (most recent call last):
  File "manage.py", line 640, in <module>
    cfg = dk.load_config(myconfig=args['--myconfig'])
  File "/home/pi/projects/donkeycar/donkeycar/config.py", line 67, in load_config
    personal_cfg.from_pyfile(personal_cfg_path)
  File "/home/pi/projects/donkeycar/donkeycar/config.py", line 19, in from_pyfile
    exec(compile(config_file.read(), filename, 'exec'), d.__dict__)
  File "/home/pi/mycar/myconfig.py", line 291, in <module>
    GYM_CONF["guid"] = "307b937a-f65b-45ee-aa3b-9f283fd7061c"
NameError: name 'GYM_CONF' is not defined

Re-enabling GYM_CONF in myconfig.py results in an error free run when running manage.py drive.

tawnkramer, 2020-09-25:
@TCIII if you remove the line 291 GYM_CONF["guid"] = "307b937a-f65b-45ee-aa3b-9f283fd7061c" it will work. Shouldn't be a problem going forward as that guid is no longer added.

closed 2020-09-25 by tawnkramer