autorope/donkeycar #627: Failed to load the native TensorFlow runtime

MrRivi, 2020-09-07:
Hi, has anyone run into these two linked errors?

1. Failed to load the native TensorFlow runtime
2. ImportError: libcuda.so.1: cannot open shared object file: No such file or directory

Thanks in advance :)

[image]
[image]

TCIII, 2020-09-07:
@MrRivi,
I assume that you have transferred your tub data and are doing the training on your PC (conda) as your Rpi can't hack it or you are attempting to train on your Nano and that requires setting up TensorRT on the Nano since it comes preinstalled with Jetpack 4.4?

MrRivi, 2020-09-07:
Correct, I am using a Raspberry pi 4B 4Gb. Following the tutorial at the donkey webpage I transferred the tub data to my PC to train the autopilot model. I am working with Ubuntu 18 on a VirtualBox but I think that shouldn't be an issue.

I also did the "Optional Install Tensorflow GPU - only for NVidia Graphics cards".

I haven't tried to train the model on the Rpi, I've read it could take a lot of time.

tawnkramer, 2020-09-07:
Don’t think the gpu cuda driver will work in a VM. Search on that and see
if I’m wrong. But try native windows install and you’ll be fine for cuda
drivers.

TCIII, 2020-09-07:
@MrRivi,
I am running conda on my HP I7 laptop with Windows 10 and haven't had a problem. As tawn said above, doing a native windows install might be better.

MrRivi, 2020-09-08:
Trying it out right now as you said, on my Dell with Windows 10. I followed the instructions for Win10 on the donkey website and no problems so far :)

The Anaconda Prompt stopped (image below) and threw out this graph. Thank you for the help!!

[image]
[image]

closed 2020-09-08 by MrRivi