autorope/donkeycar #628: Steering commands are reversed when using Web Controller

TCIII, 2020-09-08:
Ran into an interesting issue when testing the Robo HAT MM1 with and without using --js. I am running DC 3.1.5.

When I run manage.py drive --js the 2WD skid steering will correctly go left and right when the R/C controller steering wheel goes left and right and forwards and backwards when the R/C throttle trigger is pulled towards the grip handle and pushed forward away from the grip handle.

However when I use manage.py drive, the 2WD skid steering will incorrectly go right when I use the web controller circular mouse pointer (joystick) to go left and vice versa, but  correctly goes forward and backwards when I push the mouse pointer circle up and down. 

The same thing occurs with my XBox gamepad and the web controller. If I push the left joystick forward and backwards, the 2WD skid steering correctly goes forward and backwards, but when I push the right joystick to the left, the 2WD skid steering incorrectly goes right and vice versa.

Since the Robo HAT MM1 correctly commands the 2WD skid  steering to go right and left tracking the R/C controller steering wheel when I run manage.py drive --js, it seems like the robohat.py code might be sending reversed steering commands to the Robo HAT MM1 when I am using the web controller interface to steer the 2WD chassis using either the mouse joystick or the XBox gamepad controller.

TCIII, 2020-09-12:
My testing has revealed that a certain section of the robohat.py steering code is incorrect.

sctse999, 2020-09-12:
Hi @TCIII , it would greatly help me to understand the issue if you can add a test case with your expected output in https://github.com/autorope/donkeycar/blob/dev/donkeycar/tests/test_robohat.py

If you are unsure how to do this, pm me in discord.

tawnkramer, 2020-09-12:
Since we are talking about that file, I just wanted to point out that it's
not testing anything at the moment. I disabled things because they were
failing when we didn't have a robohat. I added a blank function in there
that we can fill out to let us know we have the hardware so we can run the
tests.

TCIII, 2020-09-12:
@sctse999,
I believe the problem lies in the robohat.py code here:
if steering > 0:
                output_steering = dk.utils.map_range(steering,
                                                     0, 1.0,
                                                     self.STEERING_MID, 1000)
            else:
                output_steering = dk.utils.map_range(steering,
                                                     -1, 0,
                                                     2000, self.STEERING_MID)

Be changed to:
if steering > 0:
                output_steering = dk.utils.map_range(steering,
                                                     0, 1.0,
                                                     self.STEERING_MID, 2000)
            else:
                output_steering = dk.utils.map_range(steering,
                                                     -1, 0,
                                                     1000, self.STEERING_MID)

wallarug, 2020-10-18:
Did you ever finish fixing this up @TCIII  or is this something I still need to look at?

TCIII, 2020-10-18:
I reversed the left and right steering commands being sent to the Robo HAT MM1 and that corrected the problem I was having with the reversed steering when using the web controller with the Robo HAT MM1.
You can make the correction if you want to, but since I am the only one who has noticed it and corrected it for my purposes, maybe you should let it go as an exercise for other users?

Ezward, 2023-02-18:
@sctse999 Can I close this?

TCIII, 2023-02-18:
@Ezward,

Yes.
TCIII

closed 2023-02-18 by TCIII