autorope/donkeycar #634: Simulator - AssertionError

wallarug, 2020-09-21:
When you are driving in the simulator.  This is on Ubuntu.  It is a very intermittent issue.

[image]

Students have been able to replicated it across a few different systems.

wallarug, 2020-10-18:
Hey is this resolved now? @sctse999

calumsbaird, 2020-10-25:
I'm still receiving this identical error on ubuntu at random intervals, working from the latest version on dev branch

sctse999, 2020-10-25:
this has been fixed on master branch. I didn't have time to port it back to dev branch yet. If you have time, could you please do it? you can refer to this PR.

https://github.com/autorope/donkeycar/pull/646

wallarug, 2020-10-25:
That's Jonathan for the heads up - that is something I can look at doing once I finish my work for tonight.  No worries 👍

closed 2021-01-26 by sctse999