autorope/donkeycar #638: Failed to Image while training model

julien250, 2020-09-30:
python ~/mycar/manage.py train --tub/Users/julien/Downloads/data/tub_8_20-09-26,/Users/julien/Downloads/data/tub_9_20-09-26 --model ./models/mypilot_2_24.h5

[Errno 2] No such file or directory: '/Users/julien/Downloads/data/tub_9_20-09-26/5298_cam-image_array_.jpg'
failed to load image: /Users/julien/Downloads/data/tub_9_20-09-26/5298_cam-image_array_.jpg

I get this for every image loaded from the tub. I am confused when there is no such file, because it gets the name from top directory itself.

cfox570, 2020-10-01:
Have you tried a simple command where you train all of your data?

cd ~/mycar
python manage.py train --model models/mypilot_2_24.h5

DocGarbanzo, 2021-01-16:
@julien250 - is this still a problem? You would normally copy the tub data to a folder (usually ~/mycar) and start the training from there, as described in the docs. I'll close this for now, if you follow the instructions and are still running into the issue, then please re-open this.

closed 2021-01-16 by DocGarbanzo