autorope/donkeycar #641: Gym-donkeycar munmap_chunk(): invalid pointer

natasak, 2020-10-03:
I am trying to setup the donkeycar with gym-donkeycar (20.9.14) as per instructions (guide/simulator). I have done all the pip install commands but when I try to run the gym-donkeycar I get core dumped (backtrace below). It seems like my graphics driver is causing problems, so any ideas how to solve it?

```
$ ./donkey_sim.x86_64 
munmap_chunk(): invalid pointer
Aborted (core dumped)
```

Here is the GDB backtrace I got:

```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff5b4d859 in __GI_abort () at abort.c:79
#2  0x00007ffff5bb83ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff5ce2285 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff5bc047c in malloc_printerr (str=str@entry=0x7ffff5ce41e0 "munmap_chunk(): invalid pointer") at malloc.c:5347
#4  0x00007ffff5bc06cc in munmap_chunk (p=<optimized out>) at malloc.c:2830
#5  0x00007ffff2169c4f in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#6  0x00007ffff22c122a in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#7  0x00007ffff219872b in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#8  0x00007ffff2199df0 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#9  0x00007ffff219a187 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#10 0x00007ffff219abb4 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#11 0x00007ffff1e4da58 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#12 0x00007ffff180e51c in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#13 0x00007ffff203b70c in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#14 0x00007ffff1b854d8 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#15 0x00007ffff1eea773 in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#16 0x00007ffff1f4290e in ?? () from /usr/lib/x86_64-linux-gnu/dri/amdgpu_dri.so
#17 0x00007ffff435b43f in ?? () from /opt/amdgpu-pro/lib/x86_64-linux-gnu/libGL.so.1
#18 0x00007ffff436c6e9 in ?? () from /opt/amdgpu-pro/lib/x86_64-linux-gnu/libGL.so.1
#19 0x00007ffff43632cb in ?? () from /opt/amdgpu-pro/lib/x86_64-linux-gnu/libGL.so.1
#20 0x00007ffff432d353 in glXChooseVisual () from /opt/amdgpu-pro/lib/x86_64-linux-gnu/libGL.so.1
#21 0x00007ffff7028c62 in X11_GL_GetVisual () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#22 0x00007ffff70285ec in X11_GL_LoadLibrary () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#23 0x00007ffff6fecbee in SDL_GL_LoadLibrary_REAL () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#24 0x00007ffff6fec843 in SDL_CreateWindow_REAL () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#25 0x00007ffff6feb017 in SDL_VideoInit_REAL () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#26 0x00007ffff6fdb68c in SDL_InitSubSystem_REAL () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#27 0x00007ffff6a5e41e in InitializeSDL() () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#28 0x00007ffff6a53da5 in PlayerMain(int, char**) () from /home/natasa/donkeycar-projects/DonkeySimLinux/UnityPlayer.so
#29 0x00007ffff5b4f0b3 in __libc_start_main (main=0x400620 <main>, argc=1, argv=0x7fffffffdcf8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdce8)
    at ../csu/libc-start.c:308
#30 0x0000000000400569 in _start ()
```
I am using `amdgpu` graphics drivers (see below).

```
$ sudo lshw -c display
  *-display                 
       description: VGA compatible controller
       product: Ellesmere [Radeon RX 470/480/570/570X/580/580X/590]
       vendor: Advanced Micro Devices, Inc. [AMD/ATI]
       physical id: 0
       bus info: pci@0000:01:00.0
       version: e7
       width: 64 bits
       clock: 33MHz
       capabilities: pm pciexpress msi vga_controller bus_master cap_list rom
       configuration: driver=amdgpu latency=0
       resources: irq:133 memory:e0000000-efffffff memory:f0000000-f01fffff ioport:e000(size=256) memory:f7d00000-f7d3ffff memory:c0000-dffff
```

sctse999, 2021-01-26:
@Maximellerbach 

Is there anything we need to follow up? If not, let's close this.

Maximellerbach, 2021-01-26:
I don't know what could be the reason for this error, maybe some Unity stuff I'm not aware of, or simply some drivers Issue.
Either case it's not related to DonkeyCar, you can close it.

@natasak is the problem fixed ? Would you want me to get a closer look at it ? If so, repost this issue on [this repo](https://github.com/tawnkramer/sdsandbox/). Also please provide the version of the sim you downloaded and maybe also some info on your operating system and stuff that might be related.
Thanks.

natasak, 2021-01-28:
The problem is fixed with a newer version of the amd drivers. Although they again broke it with incompatibility with higher versions of kernel. I am now on regular open source drivers and simulation is working. Operating system was Ubuntu 20.04 on x86_64. 

We can close it, although I was still able to reproduce it in case anyone would want to debug it..

closed 2021-01-28 by natasak