autorope/donkeycar #653: JSON Parse Error in tub_v2 system

wallarug, 2020-10-17:
Donkey V4
```
Traceback (most recent call last):
  File "drive.py", line 622, in <module>
    meta=args['--meta'])
  File "drive.py", line 587, in drive
    tub_writer = TubWriter(tub_path, inputs=inputs, types=types, metadata=meta)
  File "c:\users\cianb\documents\repos\wallarug\donkeycar\donkeycar\parts\tub_v2.py", line 107, in __init__
    self.tub = Tub(base_path, inputs, types, metadata, max_catalog_len)
  File "c:\users\cianb\documents\repos\wallarug\donkeycar\donkeycar\parts\tub_v2.py", line 25, in __init__
    read_only=read_only)
  File "c:\users\cianb\documents\repos\wallarug\donkeycar\donkeycar\parts\datastore_v2.py", line 232, in __init__
    self._read_contents()
  File "c:\users\cianb\documents\repos\wallarug\donkeycar\donkeycar\parts\datastore_v2.py", line 291, in _read_contents
    self.manifest_metadata = json.loads(self.seekeable.readline())
  File "C:\Program Files\ProgrammingLanguages\Python36\lib\json\__init__.py", line 354, in loads
    return _default_decoder.decode(s)
  File "C:\Program Files\ProgrammingLanguages\Python36\lib\json\decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Program Files\ProgrammingLanguages\Python36\lib\json\decoder.py", line 355, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 1 column 32 (char 31)
```

wallarug, 2020-10-17:
Fixed by deleting the manifest.json file.  Must have corrupted during testing.

DocGarbanzo, 2020-10-17:
@wallarug - whats the latest on this now? Working or not?

wallarug, 2020-10-17:
It is currently not working.

I can recreate this issue when I am using the simulator on Windows 10.   After a successful and clean exit, I cannot launch the simulator / donkey car system again without deleting manifest.json from the data folder.

I'm chatting with rahul on discord at the moment.

tikurahul, 2020-10-18:
Fixed by: https://github.com/autorope/donkeycar/commit/0789cab88824af1213a65e5768a431fa8de64dfb

Learnt something about line endings in Python today 😄

closed 2020-10-18 by tikurahul