autorope/donkeycar #680: Training model issue

Weirdeight9, 2020-11-08:
Hello, I get the following errors when I try to train my model. Could anyone help me with this?

Traceback (most recent call last):
  File "/Users/macbookpro/mycar/manage.py", line 669, in <module>
    multi_train(cfg, dirs, model, transfer, model_type, continuous, aug)
  File "/Users/macbookpro/mycar/train.py", line 902, in multi_train
    train_fn(cfg, tub, model, transfer, model_type, continuous, aug)
  File "/Users/macbookpro/mycar/train.py", line 352, in train
    records = gather_records(cfg, tub_names, opts, verbose=True)
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 439, in gather_records
    tubs = gather_tubs(cfg, tub_names)
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 419, in gather_tubs
    tubs = [Tub(p) for p in tub_paths]
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 419, in <listcomp>
    tubs = [Tub(p) for p in tub_paths]
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/parts/datastore.py", line 55, in __init__
    self.meta = json.load(f)
  File "/opt/miniconda3/envs/donkey/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/opt/miniconda3/envs/donkey/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/opt/miniconda3/envs/donkey/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/opt/miniconda3/envs/donkey/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Weirdeight9, 2020-11-08:
Update, I realized I was trying to use all of the tubs when most didn't have data in them. When I chose the one that did this was what I got.

Traceback (most recent call last):
  File "/Users/macbookpro/mycar/manage.py", line 669, in <module>
    multi_train(cfg, dirs, model, transfer, model_type, continuous, aug)
  File "/Users/macbookpro/mycar/train.py", line 902, in multi_train
    train_fn(cfg, tub, model, transfer, model_type, continuous, aug)
  File "/Users/macbookpro/mycar/train.py", line 352, in train
    records = gather_records(cfg, tub_names, opts, verbose=True)
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 439, in gather_records
    tubs = gather_tubs(cfg, tub_names)
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 419, in gather_tubs
    tubs = [Tub(p) for p in tub_paths]
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/utils.py", line 419, in <listcomp>
    tubs = [Tub(p) for p in tub_paths]
  File "/Users/macbookpro/seniordesign/donkeycar/donkeycar/parts/datastore.py", line 97, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check your tub path or provide meta info to create a new tub.

Weirdeight9, 2020-11-08:
@

closed 2020-11-08 by Weirdeight9

qasimalik134, 2021-04-13:
@Weirdeight9  Could you please share the dataset?