autorope/donkeycar #683: Errors when trying to run trained model

Weirdeight9, 2020-11-08:
Hello, I trained a model on my laptop and uploaded it on to my car. When I try to run it these are the errors I get. Can anyone help me?

Traceback (most recent call last):
  File "manage.py", line 648, in <module>
    meta=args['--meta'])
  File "manage.py", line 383, in drive
    load_model(kl, model_path)
  File "manage.py", line 348, in load_model
    kl.load(model_path)
  File "/home/pi/projects/donkeycar/donkeycar/parts/keras.py", line 51, in load
    self.model = keras.models.load_model(model_path, compile=False)
  File "/home/pi/env/lib/python3.7/site-packages/tensorflow_core/python/keras/saving/save.py", line 137, in load_model
    return hdf5_format.load_model_from_hdf5(filepath, custom_objects, compile)
  File "/home/pi/env/lib/python3.7/site-packages/tensorflow_core/python/keras/saving/hdf5_format.py", line 160, in load_model_from_hdf5
    model_config = json.loads(model_config.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'

closed 2020-11-09 by Weirdeight9

kirstenrichardson, 2020-11-10:
@Weirdeight9 did you find solution to this? Did you need to roll back the h5py version?

sctse999, 2020-11-10:
Hi, you can fix this by referring to this PR. 

We are going to merge this soon.

kirstenrichardson, 2020-11-10:
cool, that's same as what I'd done manually. Thanks