autorope/donkeycar #686: Current installation process described by online document of "Install Donkeycar on Mac" stopped working

mhchen124, 2020-11-19:
The installation steps no longer work on Mac if a user try to install Donkeycar host sw on Mac.
The command "conda env create -f install/envs/mac.yml" will generate tons of conflict message and finally fail. Discussed with Jonathan T. and he confirmed it and figured out it is caused by pinning the version of h5py package.

Expected to be fixed in a quick PR.

- Michael

sctse999, 2020-11-19:
For the record, this is the output if the version 

```
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                       |                                                                   

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package futures conflicts for:
pylint -> isort[version='>=4.2.5,<6'] -> futures
matplotlib -> tornado -> futures
tornado -> futures
tensorflow==1.13.1 -> tensorboard[version='>=1.13.0,<1.14.0'] -> futures[version='>=3.1.1']

Package enum34 conflicts for:
pylint -> astroid[version='>=1.6,<2.0'] -> enum34
tensorflow==1.13.1 -> tensorflow-base==1.13.1=mkl_py27hc36dc97_0 -> enum34

Package libpng conflicts for:
matplotlib -> libpng[version='>=1.6.32,<1.7.0a0|>=1.6.34,<1.7.0a0|>=1.6.35,<1.7.0a0|>=1.6.36,<1.7.0a0|>=1.6.37,<1.7.0a0']
pillow -> freetype[version='>=2.10.4,<3.0a0'] -> libpng[version='>=1.6.32,<1.7.0a0|>=1.6.34,<1.7.0a0|>=1.6.37,<1.7.0a0']
opencv -> libpng[version='>=1.6.32,<1.7.0a0']

Package numpy-base conflicts for:
matplotlib -> numpy[version='>=1.14.6,<2.0a0'] -> numpy-base[version='1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.4|1.14.4|1.14.4|1.14.4|1.14.4|1.14.4|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.3|1.15.3|1.15.3|1.15.3|1.15.3|1.15.3|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.2|1.16.2|1.16.2|1.16.2|1.16.2|1.16.2|1.16.3|1.16.3|1.16.3|1.16.3|1.16.3|1.16.3|1.16.4|1.16.4|1.16.4|1.16.4|1.16.4|1.16.4|1.16.5|1.16.5|1.16.5|1.16.5|1.16.5|1.16.5|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.17.2.*|1.17.3.*|1.17.4.*|1.18.1.*|1.18.5.*|1.19.1|1.19.1|1.19.1|1.19.1|1.19.1|1.19.1|1.19.2|1.17.0|1.17.0|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|>=1.9.3,<2.0a0',build='py27ha9ae307_6|py37h9797aa9_6|py36h9797aa9_6|py37h9797aa9_7|py35h9797aa9_7|py36ha9ae307_7|py27h9797aa9_7|py36ha9ae307_7|py37ha9ae307_8|py37h9797aa9_8|py35ha9ae307_8|py37he97cb71_9|py35h8a80b8c_9|py35he97cb71_9|py27h42e5f7b_10|py35h42e5f7b_10|py27h8a80b8c_10|py35h8a80b8c_10|py36h42e5f7b_10|py27ha711998_10|py35ha711998_10|py37ha711998_10|py36ha711998_10|py37ha711998_11|py27ha711998_11|py36h6575580_12|py27h6575580_12|py37ha711998_12|py27ha711998_12|py37h6575580_12|py38h6575580_13|py38h6575580_14|py38ha711998_15|py38h6575580_15|py35h479e554_1|py27h7ef55bc_1|py36ha9ae307_0|py35ha9ae307_0|py36ha9ae307_0|py36h9797aa9_0|py36h9797aa9_1|py37ha9ae307_1|py37h9797aa9_2|py36h9797aa9_2|py36ha9ae307_2|py27h9797aa9_2|py27ha9ae307_2|py27h9797aa9_3|py36h9797aa9_3|py27ha9ae307_3|py27h9797aa9_4|py36h9797aa9_4|py37ha9ae307_4|py37h9797aa9_4|py35h9797aa9_4|py35ha9ae307_4|py36h6575580_0|py36ha711998_4|py35ha711998_4|py37ha711998_4|py37h8a80b8c_4|py38ha711998_4|py38h6575580_4|py36ha711998_5|py27ha711998_5|py36h6575580_5|py37ha711998_5|py27he97cb71_0|py27h8a80b8c_0|py37he97cb71_0|py36he97cb71_0|py35he97cb71_0|py36h8a80b8c_0|py35h42e5f7b_0|py36h42e5f7b_0|py27h8a80b8c_0|py27ha711998_0|py35ha711998_0|py37ha711998_0|py37h8a80b8c_0|py37ha711998_0|py35ha711998_0|py37ha711998_1|py36ha711998_1|py36h8a80b8c_1|py37ha711998_0|py27ha711998_0|py27h8a80b8c_0|py36h8a80b8c_0|py37h8a80b8c_0|py36h8a80b8c_0|py37ha711998_0|py36h6575580_0|py37ha711998_0|py27h6575580_0|py36h6575580_0|py27ha711998_1|py37h6575580_1|py36ha711998_1|py37ha711998_1|py36h6575580_1|py37ha711998_0|py37h6575580_0|py36ha711998_1|py37h6575580_1|py27ha711998_1|py37ha711998_1|py36h6575580_0|py37ha711998_0|py37h6575580_0|py36ha711998_0|py36ha711998_0|py37ha711998_0|py36h6575580_0|py36h6575580_0|py27h6575580_0|py37ha711998_0|py36h6575580_0|py36h68fea81_0|py38h68fea81_0|py37hcfb5961_0|py37h68fea81_0|py36hcfb5961_0|py37hcfb5961_0|py37h68fea81_0|py36h68fea81_0|py38h68fea81_0|py38hcfb5961_0|py36hcfb5961_0|py38hcfb5961_0|py27h6575580_0|py27ha711998_0|py38ha711998_0|py38h6575580_0|py37h6575580_0|py36ha711998_0|py37h6575580_0|py27ha711998_0|py36ha711998_0|py37ha711998_0|py37h6575580_0|py27h6575580_0|py27ha711998_0|py36h6575580_0|py27h6575580_0|py37ha711998_0|py27ha711998_0|py37h6575580_0|py27h6575580_0|py27ha711998_0|py36ha711998_0|py36h6575580_1|py27h6575580_1|py27h6575580_0|py36h6575580_0|py27ha711998_0|py36ha711998_0|py27h6575580_1|py36ha711998_0|py37h6575580_0|py27ha711998_0|py27h6575580_0|py37h6575580_0|py36ha71 [... cut ...]
pylint -> backports.functools_lru_cache -> backports

Package pillow conflicts for:
matplotlib -> matplotlib-base[version='>=3.3.2,<3.3.3.0a0'] -> pillow[version='>=6.2.0']
pillow

Package python_abi conflicts for:
matplotlib -> setuptools -> python_abi[version='3.6.*|3.7.*',build='*_cp36m|*_cp37m']
pytest -> setuptools[version='>=40.0'] -> python_abi[version='3.6.*|3.7.*',build='*_cp36m|*_cp37m']
pip -> setuptools -> python_abi[version='3.6.*|3.7.*',build='*_cp36m|*_cp37m']

Package tornado conflicts for:
tornado
matplotlib -> tornado
```

sctse999, 2020-11-19:
Fixed in #687

closed 2020-11-19 by sctse999