autorope/donkeycar #690: Real execution cycle on embedded board is considerably lower than 20Hz and changes between drive and algo modes

cloud-rocket, 2020-11-29:
* My own setup runs at 8hz on Jetson Nano in `drive` mode and drops to 3hz in `local_angle` mode.
* Algorithm performance is probably suffers, because training and runtime is done on different frequency
* Performance was monitored via the following code - https://github.com/autorope/donkeycar/pull/689

This is my setup:
```
+--------------------+--------+--------+--------+--------+--------+--------+--------+
|        part        |  max   |  min   |  avg   |  50%   |  90%   |  99%   | 99.9%  |
+--------------------+--------+--------+--------+--------+--------+--------+--------+
|     CSICamera      |  0.09  |  0.01  |  0.02  |  0.01  |  0.02  |  0.05  |  0.09  |
| MySerialController |  0.09  |  0.02  |  0.02  |  0.02  |  0.03  |  0.06  |  0.08  |
|   ThrottleFilter   |  0.10  |  0.01  |  0.02  |  0.01  |  0.02  |  0.05  |  0.10  |
|   PilotCondition   |  0.08  |  0.01  |  0.01  |  0.01  |  0.02  |  0.04  |  0.07  |
|   RecordTracker    |  0.07  |  0.01  |  0.01  |  0.01  |  0.02  |  0.03  |  0.06  |
|        IMU         |  0.07  |  0.02  |  0.02  |  0.02  |  0.03  |  0.06  |  0.07  |
|       WebFpv       |  0.04  |  0.01  |  0.02  |  0.02  |  0.03  |  0.03  |  0.04  |
|    PerfMonitor     | 13.45  |  0.92  |  2.35  |  1.33  |  5.27  | 11.13  | 13.20  |
|     DriveMode      |  0.08  |  0.02  |  0.03  |  0.03  |  0.04  |  0.07  |  0.08  |
|      AiLaunch      |  0.04  |  0.02  |  0.02  |  0.02  |  0.02  |  0.04  |  0.04  |
|   AiRunCondition   |  0.05  |  0.01  |  0.02  |  0.01  |  0.02  |  0.02  |  0.05  |
|    PWMSteering     |  0.08  |  0.02  |  0.03  |  0.03  |  0.04  |  0.05  |  0.07  |
|    PWMThrottle     |  0.08  |  0.02  |  0.02  |  0.02  |  0.02  |  0.07  |  0.08  |
|      OLEDPart      | 155.06 | 125.55 | 136.43 | 137.56 | 142.47 | 147.86 | 154.13 |
|     Telemetry      | 15.27  |  0.11  |  0.91  |  0.12  |  4.99  | 12.03  | 14.92  |
+--------------------+--------+--------+--------+--------+--------+--------+--------+
```

sctse999, 2020-11-30:
It looks like OLEDPart is taking a lot of time to finish. Can you turn off those time consuming parts and verify the result?

cloud-rocket, 2020-12-01:
@sctse999 - I disable oled and reduced the overhead of Telemetry and PerfMonitor. I am still getting around 4hz with active model:
```
+--------------------+---------+--------+--------+--------+--------+--------+---------+
|        part        |   max   |  min   |  avg   |  50%   |  90%   |  99%   |  99.9%  |
+--------------------+---------+--------+--------+--------+--------+--------+---------+
|     CSICamera      |   0.24  |  0.01  |  0.02  |  0.02  |  0.04  |  0.05  |   0.15  |
| MySerialController |   0.10  |  0.02  |  0.03  |  0.02  |  0.03  |  0.04  |   0.09  |
|   ThrottleFilter   |   0.10  |  0.01  |  0.02  |  0.02  |  0.02  |  0.04  |   0.08  |
|   PilotCondition   |   0.08  |  0.01  |  0.01  |  0.01  |  0.02  |  0.02  |   0.07  |
|   RecordTracker    |   0.08  |  0.01  |  0.01  |  0.01  |  0.02  |  0.02  |   0.07  |
|        IMU         |   2.59  |  0.02  |  0.03  |  0.03  |  0.03  |  0.05  |   0.18  |
|       WebFpv       |   0.84  |  0.01  |  0.02  |  0.02  |  0.03  |  0.03  |   0.05  |
|    PerfMonitor     |   0.24  |  0.02  |  0.03  |  0.03  |  0.04  |  0.06  |   0.11  |
|    FileWatcher     |  18.39  |  0.05  |  0.25  |  0.16  |  0.29  |  3.76  |   7.49  |
|    FileWatcher     |   9.76  |  0.03  |  0.17  |  0.08  |  0.20  |  3.24  |   6.40  |
|   DelayedTrigger   |   0.09  |  0.01  |  0.02  |  0.02  |  0.02  |  0.04  |   0.08  |
| TriggeredCallback  |   0.09  |  0.01  |  0.01  |  0.01  |  0.01  |  0.02  |   0.09  |
|    KerasLinear     | 1515.96 | 289.63 | 329.83 | 321.82 | 344.91 | 386.60 | 1115.74 |
|     DriveMode      |   0.24  |  0.02  |  0.04  |  0.04  |  0.10  |  0.13  |   0.24  |
|      AiLaunch      |   0.08  |  0.01  |  0.02  |  0.02  |  0.02  |  0.04  |   0.08  |
|   AiRunCondition   |   0.09  |  0.01  |  0.01  |  0.01  |  0.02  |  0.03  |   0.07  |
|    PWMSteering     |   0.60  |  0.02  |  0.07  |  0.07  |  0.11  |  0.16  |   0.22  |
|    PWMThrottle     |   0.17  |  0.02  |  0.03  |  0.03  |  0.04  |  0.05  |   0.12  |
|   MqttTelemetry    |   0.94  |  0.09  |  0.13  |  0.13  |  0.14  |  0.19  |   0.38  |
+--------------------+---------+--------+--------+--------+--------+--------+---------+
```

tikurahul, 2020-12-01:
You should try using TensorRT on the Nano or TFLite. Your framerate will improve considerably. I usually get 45Hz with my model.
A p90 of `344.91`ms is actually _vey_ poor.

DocGarbanzo, 2021-01-16:
@cloud-rocket - if you still see the problem, please re-open this issue with more info. In particular if you run the script `profile.py` with all 3 models: h5, rt, tflite and are not getting a resolution from that? I'll close this for now as I think TensorRT is the right format for the Nano.

closed 2021-01-16 by DocGarbanzo

cloud-rocket, 2021-01-20:
I am not yet reopening, but `tflite` performance I am getting is 19hz (just for statistics)

sctse999, 2021-01-20:
@cloud-rocket  This is Nano 2GB?

cloud-rocket, 2021-01-20:
Nop - 4Gb.

I added np.float32 conversion to `tflite.py` part, because otherwise it was throwing exception (the image was called with float64 values). Maybe because I am using TF2.4....
```
    def inference(self, img_arr, other_arr):
        input_data = np.float32(img_arr.reshape(self.input_shape))
        self.interpreter.set_tensor(self.input_details[0]['index'], input_data)
        self.interpreter.invoke()
```

DocGarbanzo, 2021-01-22:
We saw problems with tf > 2.2.0 in training. But 20 Hz is definitely much too slow. Have you tried the CPU version of TF?

cloud-rocket, 2021-01-22:
@DocGarbanzo, WDYM by "CPU version"? With standard TF I am getting 3.5hz - this is why I started this ticket....

BTW added a PR for tflite fix - https://github.com/autorope/donkeycar/pull/762

Also trying to see how to run TensorRT with TF2 - many changes required...

DocGarbanzo, 2021-01-31:
@cloud-rocket - I just looked at the issue w/ TF > 2.2.0 on RPi yesterday. Indeed, any newer version causes a significant slowness on RPi, at least when using the `.h5` version of the model. On a RPi 4 I saw around 180ms using TF 2.3.1 vs 30ms using TF 2.2.0. So we updated RPi install instructions on dev. With the `tflite` model it drops to ~20ms.