autorope/donkeycar #692: Why isn't the T265 configuration available in /templates/complete.py

TCIII, 2020-11-30:
I am running DC 3.1.5 on a NVIDIA Nano 4GB and would like to integrate my Realsense T265 into the DC software. However, at the moment there appears to be a CAMERA_TYPE only available for the Realsense D435i in /templates/complete.py.

All of the DC python support programs (/templates/path_follow.py, /templates/cfg_path_follow.py, /parts/realsense2.py, and /parts/path.py) appear to be available to support the T265.

Is there a way to configure the DC configuration to use the Realsense T265?

TCIII, 2020-12-06:
I have found that a version of DC can be built for the Realsense T265 tracking camera using the template "path_follow". However the "manage.py" code that is built attempts to import "pigpio" which does not exist for DC when built for the NVIDIA Nano instead it is jetson-gpio. 
I downloaded the DC master files and used "pip install -e .[nano]" to install the Nano version of the files and not the Rpi version.
Comments?

Ezward, 2023-03-04:
T265 is not longer suppored by Intel, so I'm going to close this issue.

closed 2023-03-04 by Ezward