autorope/donkeycar #700: "  python manage.py drive "    has error

xiaohai-AI, 2020-12-14:
Jetson nano when I run the above command the following is what I got:

using donkey v4.0.1-dev ...
loading config file: /home/jetrobcar/mycar/config.py
loading personal config over-rides from myconfig.py
Adding part CSICamera.
Starting Donkey Server...... you can now go to jetrobcar-desktop.local:8887 to drive your car.
Adding part LocalWebController.
Adding part PilotCondition.
Adding part DriveMode.
Traceback (most recent call last):
  File "manage.py", line 171, in <module>
    drive(cfg, model_path=args['--model'], model_type=args['--type'])
  File "manage.py", line 136, in drive
    if cfg.DONKEY_GYM or cfg.DRIVE_TRAIN_TYPE == "MOCK":
AttributeError: 'Config' object has no attribute 'DRIVE_TRAIN_TYPE'

EricWiener, 2020-12-24:
Could you post what your `config.py` is? There is a `DRIVE_TRAIN_TYPE` in the default complete config.

xiaohai-AI, 2020-12-26:
This is what I found, if i execute the following command for 4.0.1 version
"donkey createcar --path ~/mycar " 
I will get the mentioned issue, but if I execute the following command
"donkey createcar --path ~/mycar --template complete"
then the probleam is gone

Hyunwoo-Park-Yonsei, 2021-01-12:
> This is what I found, if i execute the following command for 4.0.1 version
> "donkey createcar --path ~/mycar "
> I will get the mentioned issue, but if I execute the following command
> "donkey createcar --path ~/mycar --template complete"
> then the probleam is gone

So you have solved the error by executing "donkey createcar --path ~/mycar --template complete" ??

xiaohai-AI, 2021-01-12:
Yes发自我的华为手机-------- 原始邮件 --------发件人： Hyunwoo-Park-Yonsei <notifications@github.com>日期： 2021年1月11日周一 晚上8:01收件人： autorope/donkeycar <donkeycar@noreply.github.com>抄送： xiaohai-AI <zhouxh90@gmail.com>, Author <author@noreply.github.com>主    题： Re: [autorope/donkeycar] "  python manage.py drive "    has error (#700)

This is what I found, if i execute the following command for 4.0.1 version
"donkey createcar --path ~/mycar "
I will get the mentioned issue, but if I execute the following command
"donkey createcar --path ~/mycar --template complete"
then the probleam is gone

So you have solved the error by executing "donkey createcar --path ~/mycar --template complete" ??

zlite, 2021-01-12:
Depending on which template you're using, this may have been fixed in [this PR](https://github.com/autorope/donkeycar/pull/725). 

In general, anytime you see that sort of error ('Config' object has no attribute .....) you can solve it by adding the relevant line from the complete config.py to your myconfig.py

DocGarbanzo, 2021-01-16:
Fixed in #725

closed 2021-01-16 by DocGarbanzo