autorope/donkeycar #703: DS4DRV no longer necessary for PS4 controller operation with Donkey Car running on Buster

TCIII, 2020-12-15:
The following guide will pair and connect the PS4 to an Rpi running Buster from the CLI without requiring ds4drv:

SSH into your Rpi and from the CLI type:

sudo bluetoothctl

agent on

default-agent

scan on

Turn on your PS4 pairing (fast blinking blue light)

PS4 MAC_ADDRESS will appear:

 [NEW] Device XX:XX:XX:XX:XX:XX Wireless Controller (where XX:XX:XX:XX:XX:XX is the controller PS4_MAC_ADDRESS)

connect PS4_MAC_ADDRESS (should result in:)

Attempting to connect to 00:01:6C:B4:06:7E
 [CHG] Device 00:01:6C:B4:06:7E Connected: yes
 [CHG] Device 00:01:6C:B4:06:7E UUIDs: 00001124-0000-1000-8000-00805f9b34fb
 [CHG] Device 00:01:6C:B4:06:7E UUIDs: 00001200-0000-1000-8000-00805f9b34fb
 [CHG] Device 00:01:6C:B4:06:7E ServicesResolved: yes
 [CHG] Device 00:01:6C:B4:06:7E Paired: yes
 Connection successful

trust PS4_MAC_ADDRESS

sudo apt-get install joystick (incase joystick is not already installed on Buster)

Now test your PS4 joystick:

sudo jstest /dev/input/js0

 Now if values change when you move and press buttons on your controller, 
 then you are ready to go with your PS4 controller, 
 and you can finish the guide here.

sctse999, 2021-01-26:
Cool. Let's put this into the doc together.

Ezward, 2023-02-18:
docs have been updated https://docs.donkeycar.com/parts/controllers/#ps4-dualshock-4-wireless-gamepad-controller

closed 2023-02-18 by Ezward