autorope/donkeycar #716: Anaconda enviroment - Found conflicts! Looking for incompatible packages.

TheNewCivilian, 2020-12-29:
I freshly installed Miniconda3 as described [here]( https://github.com/autorope/donkeycar/edit/master/docs/guide/host_pc/setup_ubuntu.md) on my Arch-linux system. During the environment creation I got this output:

```
Collecting package metadata (repodata.json): done
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                              -  
Solving environment: - 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                 

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package numpy conflicts for:
h5py==2.10.0 -> numpy[version='>=1.11.3,<2.0a0|>=1.14.6,<2.0a0|>=1.16.6,<2.0a0']
tensorflow==1.13.1 -> tensorboard[version='>=1.13.0,<1.14.0'] -> numpy[version='>=1.12.0|>=1.13.3,<2.0a0|>=1.14.6,<2.0a0']
pandas -> numpy[version='>=1.11.3,<2.0a0|>=1.12.1,<2.0a0|>=1.13.3,<2.0a0|>=1.14.6,<2.0a0|>=1.15.4,<2.0a0|>=1.16.6,<2.0a0|>=1.9.3,<2.0a0|>=1.9']
matplotlib -> matplotlib-base[version='>=3.3.2,<3.3.3.0a0'] -> numpy[version='>=1.15.4,<2.0a0|>=1.16.6,<2.0a0']
matplotlib -> numpy[version='>=1.14.6,<2.0a0']
opencv -> py-opencv==3.4.2=py37h765d7f9_1 -> numpy[version='>=1.11.3,<2.0a0']
opencv -> numpy[version='>=1.9.3,<2.0a0']

Package tzdata conflicts for:
pip -> python[version='>=3.9,<3.10.0a0'] -> tzdata
pylint -> python[version='>=3.9,<3.10.0a0'] -> tzdata
tornado -> python[version='>=3.9,<3.10.0a0'] -> tzdata
pillow -> python[version='>=3.9,<3.10.0a0'] -> tzdata
h5py==2.10.0 -> python[version='>=3.9,<3.10.0a0'] -> tzdata
pytest -> python[version='>=3.9,<3.10.0a0'] -> tzdata
pandas -> python[version='>=3.9,<3.10.0a0'] -> tzdata

Package pip conflicts for:
pip
tensorflow==1.13.1 -> python=2.7 -> pip
pytest -> python[version='>=3.8,<3.9.0a0'] -> pip
pylint -> python[version='>=3.9,<3.10.0a0'] -> pip
opencv -> python[version='>=3.7,<3.8.0a0'] -> pip
docopt -> python[version='>=3.8,<3.9.0a0'] -> pip
pandas -> python[version='>=3.6,<3.7.0a0'] -> pip
pillow -> python[version='>=3.9,<3.10.0a0'] -> pip
h5py==2.10.0 -> python[version='>=3.8,<3.9.0a0'] -> pip
tornado -> python[version='>=3.6,<3.7.0a0'] -> pip
matplotlib -> python[version='>=3.6,<3.7.0a0'] -> pip

Package ca-certificates conflicts for:
pandas -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
pytest -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
pip -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
opencv -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
pillow -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
tensorflow==1.13.1 -> python=2.7 -> ca-certificates
matplotlib -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
tornado -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
docopt -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
pylint -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates

Package zstd conflicts for:
opencv -> libtiff[version='>=4.0.9,<5.0a0'] -> zstd[version='>=1.3.7,<1.3.8.0a0|>=1.4|>=1.4.4,<1.5.0a0']
pillow -> libtiff[version='>=4.1.0,<5.0a0'] -> zstd[version='>=1.3.7,<1.3.8.0a0|>=1.4|>=1.4.4,<1.5.0a0']

Package _libgcc_mutex conflicts for:
tornado -> libgcc-ng[version='>=7.3.0'] -> _libgcc_mutex=[build=main]
matplotlib -> libgcc-ng[version='>=7.3.0'] -> _libgcc_mutex=[build=main]
h5py==2.10.0 -> libgcc-ng[version='>=7.3.0'] -> _libgcc_mutex=[build=main]
pandas -> libgcc-ng[version='>=7.3.0'] -> _libgcc_mutex=[build=main]
opencv -> libgcc-ng[version='>=7.2.0'] -> _libgcc_mutex=[build=main]
[... 73 lines cut ...]
pandas -> numpy[version='>=1.16.6,<2.0a0'] -> numpy-base[version='1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.11.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.3|1.14.4|1.14.4|1.14.4|1.14.4|1.14.4|1.14.4|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.5|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.14.6|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.0|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.1|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.2|1.15.3|1.15.3|1.15.3|1.15.3|1.15.3|1.15.3|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.15.4|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.0|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.1|1.16.2|1.16.2|1.16.2|1.16.2|1.16.2|1.16.2|1.16.3|1.16.3|1.16.3|1.16.3|1.16.3|1.16.3|1.16.4|1.16.4|1.16.4|1.16.4|1.16.4|1.16.4|1.16.5|1.16.5|1.16.5|1.16.5|1.16.5|1.16.5|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.16.6|1.17.2.*|1.17.3.*|1.17.4.*|1.18.1.*|1.18.5.*|1.19.1|1.19.1|1.19.1|1.19.1|1.19.1|1.19.1|1.19.2|1.17.0|1.17.0|1.17.0|1.17.0|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|1.9.3|>=1.9.3,<2.0a0',build='py36h2b20989_6|py37hdbf6ddf_6|py36hdbf6ddf_6|py36hdbf6ddf_7|py36h2b20989_7|py35hdbf6ddf_7|py35h2b20989_7|py37hdbf6ddf_7|py36h2b20989_7|py27hdbf6ddf_7|py36hdbf6ddf_7|py36h2b20989_8|py37h2b20989_8|py35hdbf6ddf_8|py36h7cdd4dd_9|py27h7cdd4dd_9|py36h3dfced4_9|py35h3dfced4_9|py35h74e8950_9|py37h74e8950_9|py36h74e8950_9|py27h81de0dd_9|py35h81de0dd_9|py35h74e8950_10|py36h74e8950_10|py27h81de0dd_10|py37h81de0dd_10|py36h81de0dd_10|py37h2f8d375_10|py36h2f8d375_10|py27h2f8d375_10|py35h2f8d375_10|py36h2f8d375_11|py36hde5b4d6_11|py27hde5b4d6_11|py37h2f8d375_12|py27h2f8d375_12|py27hde5b4d6_12|py36hde5b4d6_12|py38hde5b4d6_12|py35h0ea5e3f_1|py27h9be14a7_1|py35hdbf6ddf_0|py27h2b20989_0|py35hdbf6ddf_0|py36hdbf6ddf_1|py37hdbf6ddf_1|py27h2b20989_1|py37h2b20989_2|py27hdbf6ddf_2|py37hdbf6ddf_2|py36h2b20989_3|py36hdbf6ddf_3|py27hdbf6ddf_4|py37h2b20989_4|py36hdbf6ddf_4|py35hdbf6ddf_4|py37h2f8d375_4|py36h2f8d375_4|py27h2f8d375_4|py37hde5b4d6_5|py36hde5b4d6_5|py36h7cdd4dd_0|py35h3dfced4_0|py37h3dfced4_0|py37h74e8950_0|py35h81de0dd_0|py36h2f8d375_0|py37h2f8d375_0|py37h2f8d375_0|py27h81de0dd_0|py35h81de0dd_0|py35h2f8d375_0|py36h2f8d375_0|py27h81de0dd_1|py37h81de0dd_1|py37h2f8d375_1|py36h2f8d375_0|py37h2f8d375_0|py27h81de0dd_0|py37h81de0dd_0|py37h2f8d375_0|py37h81de0dd_0|py27h81de0dd_0|py37h2f8d375_0|py36h2f8d375_0|py37h2f8d375_1|py27hde5b4d6_1|py37h2f8d375_0|py36h2f8d375_0|py36hde5b4d6_0|py37h2f8d375_1|py36hde5b4d6_1|py36h2f8d375_0|py36h2f8d375_0|py37hde5b4d6_0|py36h2f8d375_0|py37h2f8d375_0|py27hde5b4d6_0|py36h2f8d375_0|py37h2f8d375_0|py37h2f8d375_0|py36h2f8d375_0|py37hde5b4d6_0|py38hde5b4d6_0|py39hfb011de_1|py38hfa32c7d_0|py36hfa32c7d_0|py37h75fe3a5_0|py37h75fe3a5_0|py39h2ae0177_0|py39h0f7b65f_0|py38hfa32c7d_0|py36hfa32c7d_0|py36h75fe3a5_0|py38h75fe3a5_0|py37hfa32c7d_0|py36h75fe3a5_0|py38h75fe3a5_0|py37hfa32c7d_0|py39h76555f2_1|py27hde5b4d6_0|py27h2f8d375_0|py38h2f8d375_0|py36hde5b4d6_0|py36hde5b4d6_0|py37hde5b4d6_0|py36hde5b4d6_0|py37hde5b4d6_0|py37h2f8d375_0|py27h2f8d375_0|py36h2f8d375_0|py36hde5b4d6_0|py27hde5b4d6_0|py37hde5b4d6_0|py27h2f8d375_0|py36hde5b4d6_0|py27hde5b4d6_0|py37h2f8d375_0|py27h2f8d375_0|py36hde5b4d6_0|py27hde5b4d6_0|py37hde5b4d6_0|py27h2f8d375_0|py37h2f8d375_0|py27hde5b4d6_1|py37hde5b4d6_1|py36h2f8d375_1|py27h2f8d375_1|py27hde5b4d6_0|py37hde5b4d6_0|py27h2f8d375_0|py36hde5b4d6_1|py37hde5b4d6_1|py36h2f8d375_1|py27h2f8d375_1|py36hde5b4d6_0|py27hde5b4d6_0|py37hde5b4d6_0|py27h2f8d375_0|py37hde5b4d6_0|py36hde5b4d6_0|py27hde5b4d6_0|py36h81de0dd_0|py27h2f8d375_0|py36h2f8d375_0|py36h81de0dd_0|py27h2f8d375_0|py36h81de0dd_1|py27h2f8d375_1|py36h2f8d375_1|py36h81de0dd_0|py37h81de0dd_0|py27h2f8d375_0|py27h2f8d375_0|py35h2f8d375_0|py37h81de0dd_0|py27h81de0dd_0|py36h81de0dd_0|py36h74e8950_ [... cut ...]
Package backports.functools_lru_cache conflicts for:
pylint -> backports.functools_lru_cache
matplotlib -> backports.functools_lru_cache
```
Adding these channels to the env file fixed it for me:
```
channels:
  - conda-forge
  - defaults
  - conda-forge/label/broken
```
I don`t know enough about anaconda environments to decide whether or not this might be a general issue. If that is the case putting a small note on the setup instructions might prevent a lot headache.
If someone gives me the go, I can open a simple pull request for it.

albus12cb11, 2020-12-31:
You basically saved my life with those 3 lines. I've been struggling with this error for the last hour.

TheNewCivilian, 2021-01-02:
Looks like this has already been fixed on dev. https://github.com/autorope/donkeycar/blob/a5c1424717ef50db00d0b2ab6c2a637971a94483/install/envs/ubuntu.yml#L3

closed 2021-01-02 by TheNewCivilian