autorope/donkeycar #719: Chaining IMU and Servo Driver

jamesdi1993, 2021-01-03:
Hello, 

I'm pretty new to electronics and circuit design, and my question might be naive. I couldn't figure out how to connect the IMU with the PI per the website instruction "Typically uses the I2C interface and can be chained off the default PWM PCA9685 board. This configuration will also provide power".

Specifically, what does chaining mean here?  As far as I know, both IMU and the PCA 9685 both use I2c buses for communication. My PI-2B only have 2 i2c pins so I can understand the necessity of sharing I2c bus here. However, the servo board has I2C as input, and should output PWB.  I'm a bit confused how to do this chaining of the IMU and the servo board because their input & output don't seem to match.  

I tried to look up some tutorials online but the resources seem a bit scarce. If you have some resources for reference that would be greatly appreciated, too. 

Thanks in advance!

DocGarbanzo, 2021-01-16:
Hi @jamesdi1993,

You can just connect the 2 devices (PCA 9685) and IMU in parallel. In fact the PCA (9685) has pins on both side, so you use one side to connect to your Pi and connect the IMU on the other side.

closed 2021-01-16 by DocGarbanzo

jamesdi1993, 2021-02-07:
Hi, 

Thanks for your reply. I tried your suggestions and can see that my IMU is registered at address 68 when command "sudo i2cdetect -y 1".

However, when I run "python manage.py drive" it throws the following error:

using donkey v4.1.0 ...
loading config file: /home/pi/mycar/config.py
loading personal config over-rides from myconfig.py
PiCamera loaded.. .warming camera
Adding part PiCamera.
Starting Donkey Server...... you can now go to pi-james.local:8887 to drive your car.
Adding part LocalWebController.
Adding part PilotCondition.
Adding part DriveMode.
Traceback (most recent call last):
  File "manage.py", line 172, in <module>
    drive(cfg, model_path=args['--model'], model_type=args['--type'])
  File "manage.py", line 141, in drive
    busnum=cfg.PCA9685_I2C_BUSNUM)
  File "/home/pi/projects/donkeycar/donkeycar/parts/actuator.py", line 30, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685(address=address)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 322, in write_byte_data
    self._device.write(data)
OSError: [Errno 121] Remote I/O error

Do you know what additional steps I need to do to enable IMU?

jamesdi1993, 2021-02-08:
I was eventually able to integrate the IMU by connecting the power and ground with the Pi, and then connecting SDA and SCL with the other side of the PCA board.