autorope/donkeycar #724: L298N class appear in config.py

Erdi17, 2021-01-04:
I'm using a motor controller hat where the required motor control inputs are virtually the same as the L298N: Two directional control logic level inputs and a PWM input (0-255) for speed control per motor. 
If you look here: https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/actuator.py 
you will find code for the L298N motor controller:  class L298N_HBridge_DC_Motor(object).

Unfortunately this L298N class does not appear in the template cfg_complete.py to make the L298N appear in config.py.

Having this class available in the config.py would make things way easier for me and hopefully for other people that use the same motor controller and the 2 wheels powered by two motors and a ball pen wheel car model

DocGarbanzo, 2021-01-16:
@Erdi17 - can you make the corresponding change in `cfg_complete.py`, test it, and submit a PR?

python1995, 2021-01-18:
@DocGarbanzo , I'm sorry but I was not able to do so, hoping someone that is facing the same problem and has more knowledge will help with this issue.

DocGarbanzo, 2021-01-22:
@Erdi17 - have you tested that part and can you confirm it works?

sctse999, 2021-01-26:
@Erdi17 If you can help verifying this part is working, I am happy to add this back to https://github.com/autorope/donkeycar/blob/dev/donkeycar/templates/complete.py

python1995, 2021-01-26:
@sctse999, I am not sure how can I help verifying if  this part is working ? I am willing to try your suggestions on my robotcar and see if it works. thank you

python1995, 2021-01-26:
@DocGarbanzo , I haven't done anything. I am not really good enogh with the donkeycar to do implement changes. what should I do to help ?

python1995, 2021-01-26:
@sctse999 I am not sure how can I help verifying if this part is working ? I am willing to try your suggestions on my robotcar and see if it works. thank you

raskaman, 2021-03-31:
added a PR for this issue:
https://github.com/autorope/donkeycar/pull/826

Ezward, 2023-02-17:
Merged PR https://github.com/autorope/donkeycar/pull/829

closed 2023-02-17 by Ezward