autorope/donkeycar #727: Conda environment file (windows.yml) for Windows somehow got dropped in Dev

zlite, 2021-01-06:
[This folder](https://github.com/autorope/donkeycar/tree/dev/install/envs) should have windows.yml as described in the docs [here](http://docs.donkeycar.com/guide/host_pc/setup_windows/)

Hyunwoo-Park-Yonsei, 2021-01-11:
Having same error. Did you find windows.yml file?

Hyunwoo-Park-Yonsei, 2021-01-11:
I have found window.yml file. 
Go to below link and download the relaese 3.1 and there is windows.yml
https://github.com/autorope/donkeycar/releases

tikurahul, 2021-01-11:
We no longer need a `windows.yml` file. Please use `ubuntu.yml` for now. That should get renamed to `pc.yaml` as there are no x-platform differences.

custom-build-robots, 2021-01-12:
@tikurahul today I tested once again the "ubuntu.yml" file on my Windows 10 laptop with conda and we hat a short discussion on discord. The problem is that tensorflow 2.2.0 is not available for conda.
If you use the ubuntu.yml installation file then you have to change the tensorflow version to 2.3.0. or 2.1.0 due to the fact that 2.2.0 is not available...

tikurahul, 2021-01-12:
We can switch to 2.3.0.

DocGarbanzo, 2021-01-16:
I've updated the `ubuntu.yml` to allow tf >= 2.2.0 <= 2.3.2. I'll do a PR to rename the file to `pc.yml` and update the docs.

DocGarbanzo, 2021-01-16:
Addressed in #748.

closed 2021-01-16 by DocGarbanzo