autorope/donkeycar #735: Having OSError: [Errno 121] Remote I/O error

Hyunwoo-Park-Yonsei, 2021-01-12:
When I type    python manage.py drive   and trying to go <your car's hostname.local>:8887 and control my car.
I got this error message.

using donkey v4.1.0-dev ...
loading config file: /home/pi/mycar/config.py
loading personal config over-rides from myconfig.py
PiCamera loaded.. .warming camera
Adding part PiCamera.
Starting Donkey Server...... you can now go to raspberrypi.local:8887 to drive your car.
Adding part LocalWebController.
Adding part PilotCondition.
Adding part DriveMode.
Traceback (most recent call last):
  File "manage.py", line 171, in <module>
    drive(cfg, model_path=args['--model'], model_type=args['--type'])
  File "manage.py", line 140, in drive
    busnum=cfg.PCA9685_I2C_BUSNUM)
  File "/home/pi/projects/donkeycar/donkeycar/parts/actuator.py", line 30, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685(address=address)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/pi/env/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 322, in write_byte_data
    self._device.write(data)
OSError: [Errno 121] Remote I/O error

TCIII, 2021-01-13:
The error you are seeing is usually caused by the DC manag.py program not being able to communicate with the PCA9685 servo controller over the I2C bus. If you are using an Rpi 3/4 did you verify that the software was seeing the PCA9685 at hex address 0x40? If not that may be your problem.
However if you are not using a PCA9685 servo controller, then you have a problem unless the servo controller performs the same way and can be addressed like the PCA9685 then you can change the servo controller address in myconfig.py.

sctse999, 2021-01-26:
Please following the instruction and use `i2cdetect` to check the connectivity of PCA9685.

http://docs.donkeycar.com/guide/create_application/#configure-i2c-pca9685

I am closing this issue now. If you are still experiencing issue, feel free to reopen it.

closed 2021-01-26 by sctse999