autorope/donkeycar #742: Epoch failure during train command

kelvham, 2021-01-13:
When the 
`python ~/mycar/train.py --tub <tub folder names comma separated> --model ./models/mypilot.h5`
command  is ran (any form of it), the epochs start but eventually fail with error
`W tensorflow/core/kernels/data/generator_dataset_op.cc:103] Error occurred when finalizing GeneratorDataset iterator: Failed precondition: Python interpreter state is not initialized. The process may be terminated.
         [[{{node PyFunc}}]]` .

tikurahul, 2021-01-14:
Which version of TensorFlow are you using ? You should be using 2.2.0 or 2.3.0.

kelvham, 2021-01-14:
> Which version of TensorFlow are you using ? You should be using 2.2.0 or 2.3.0.

I am using 2.3.0

tikurahul, 2021-01-14:
Can you please do the following:

```
import tensorflow as tf
print(tf.__version__)
```

By opening a Python REPL in your environment. I have only seen this happen when using TF 1.15.

kelvham, 2021-01-14:
I'm sorry, I am brand new to Python, I think this is what you were looking for:

`(donkey) C:\Users\Kelvin>python
Python 3.7.9 (default, Aug 31 2020, 17:10:11) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow as tf
>>> print(tf.__version__)
2.3.0`

tikurahul, 2021-01-14:
I can see your virtualenv activated here. What happens when you try training now ?
Can you send us all the logs in the console ? 

I have a feeling that you might be running out of memory so maybe try decreasing the batch size ?
Related: https://github.com/tensorflow/tensorflow/issues/24570

kelvham, 2021-01-14:
@tikurahul this is the consoles output when I try training 
[console.txt](https://github.com/autorope/donkeycar/files/5816250/console.txt).

I actually was trying to run it with a smaller batch size but I could not find where/what file to edit to decrease the batch. I am brand new to all this stuff, so I am not sure what resources of the PC it uses, but I have an i7 8700k, GTX 1080, and 32 GB of RAM, so I did not think I would reach any limitations.

tikurahul, 2021-01-16:
Sorry, I am not sure what is going on in your case. I recommend opening a bug in the TF repo.

DocGarbanzo, 2021-01-16:
Can you try w/ latest TensorFlow 2.3.2: `pip install tensorflow==2.3.2`? Btw, you are not using your graphics card for training, you'd need to install `tensorflow-gpu` instead. But I suggest you fix the current problem first as the gpu setup can cause a lot of other issues. Memory-wise you should be absolutely fine.

kelvham, 2021-01-17:
@DocGarbanzo not sure exactly what went wrong here, but ended up with the same error.
[console with 2.3.2.txt](https://github.com/autorope/donkeycar/files/5825515/console.with.2.3.2.txt)

tikurahul, 2021-01-17:
With TF 2.x there is no `tensorflow-gpu`. Its all the same package.

Can you please try using the latest stable version ? That's `TF 2.4.0`. 
https://www.tensorflow.org/versions has more info.

kelvham, 2021-01-17:
@tikurahul Same issue with 2.4.0
[console with 2.4.0.txt](https://github.com/autorope/donkeycar/files/5825560/console.with.2.4.0.txt)

tikurahul, 2021-01-17:
At this point I would recommend filing a bug with TF. I have a feeling that might be fastest way to get to the bottom of this.

custom-build-robots, 2021-01-20:
I switched back from TensorFlow 2.3.0 to TensorFlow 2.1.0 in my Windows Anaconda installation. Now I am getting a new error attached as picture.
The error is repeatable on different computers and different Anaconda installations. I used always the ubuntu.yml installation file and only changed the version for TensorFlow to `tensorflow==2.1.0`.
Not sure if I should open a new issue?
[image]

DocGarbanzo, 2021-01-22:
Can you please try and install TensorFlow 2.2.0 - that version is in the conda files and the setup file and it is the version that is tested and proven to work on OSX and Linux.

sctse999, 2021-01-26:
perhaps conda doesn't offer Tensorflow 2.2.0 on windows?

https://anaconda.org/anaconda/tensorflow/files?version=2.2.0&page=2

@custom-build-robots are you able to install TF 2.2.0 on your platform?

kelvham, 2021-01-26:
@DocGarbanzo @sctse999 I was using 2.2.0, but was told to upgrade.

sctse999, 2021-01-26:
> > Which version of TensorFlow are you using ? You should be using 2.2.0 or 2.3.0.
> 
> I am using 2.3.0 @tikurahul

Were you using 2.2.0 or 2.3.0?

kelvham, 2021-01-26:
@sctse999 I was originally using 2.2.0, as default in the yml file. But, as you suspected, conda does not support 2.2.0, so I had to upgrade.

DocGarbanzo, 2021-01-26:
@kelvham - if there is no conda package for tf 2.2.0 under windows, then please install tf 2.2.0 with pip. We have to do the same thing on OSX. Which also means our `pc.yml` file needs to be split again into `ubuntu.yml` and `win.yml`.

kelvham, 2021-01-26:
@DocGarbanzo Sorry I may not completely understand how it all works, but I did install tf 2.2.0 using pip at one point yielding the same results.  I believe we talked about it over the Discord server which is why it’s not documented here. Nonetheless, I am going to try to run it with 2.2.0 again when I get the chance.

kelvham, 2021-01-28:
@DocGarbanzo, it looks like, it may have worked? No error message, but it stopped at epoch 39, is that normal?
[console with 2.2.0.txt](https://github.com/autorope/donkeycar/files/5888716/console.with.2.2.0.txt)

I copied my model file back to the car but when I try driving the car based on the model, I get another error.
[donkey model fail.txt](https://github.com/autorope/donkeycar/files/5888729/donkey.model.fail.txt)

DocGarbanzo, 2021-01-31:
Yep, training worked. On the RPi you need to install tf 2.2.0 as well. Our current docs (docs.donkeycar.com) unfortunately show the installation for tf 2.3.1. Please look up the docs under the `dev` branch - we just updated them yesterday - they contain the RPi install for 2.2.0 now.

closed 2021-01-31 by kelvham

mf01, 2021-02-08:
I can confirm this bug using DonkyCar 4.1.0 with Pythonh 3.8.5 and TensorFlow 2.4.1 on Win10.

**`Epoch 00014: val_loss did not improve from 0.00264
2021-02-08 10:24:48.692570: W tensorflow/core/kernels/data/generator_dataset_op.cc:107] Error occurred when finalizing GeneratorDataset iterator: Failed precondition: Python interpreter state is not initialized. The process may be terminated.
         [[{{node PyFunc}}]]`**